prusa3d/Prusa-Firmware-Buddy #1580: [ENHANCEMENT] Adjust Filament loading/unloading speed with Bondtech extruder

linx25, 2021-09-08:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
Printer type - MINI+
Firmware - 4.3.3-RC

Is your feature request related to a problem? Please describe.
Loading of the FIlament make trouble when there is a Bondtech extruder mounted

Describe the solution you'd like
LCD Menu -> Settings -> HW settings -> then long-press the knob
Please add the option to adjust the loading and unloading  speed 
or a option to set it via gcode and the option to save it.

This improvement would be very helpful and would avoid harm for some MINI+ users :)

BastR, 2021-09-11:
I confirm that I have the same problem, the stepper E stalls when loading/unloading. I have to reduce the pressure on the lever to make it work.

I don't know if it's a speed problem or if the E current is too low?

JohnnyDeer, 2021-09-16:
Hi @linx25 @BastR, we are not experiencing any stalling or grinding. Can you please describe this behavior more and ideally attach the video? Thank you.

BastR, 2021-09-16:
Hi @JohnnyDeer, here the video:

https://user-images.githubusercontent.com/20129420/133644472-b295a567-c8f8-4454-8357-2dc95dd396e1.mp4

As I said in my previous message, if I loosen the screw on the extruder door or if I help the filament to come out by hand, I can eject it.

linx25, 2021-09-16:
same here exact the same problem

edwardhorsford, 2021-09-16:
Not experiencing the issue here - though using startup gcode to set the esteps in advance.

It would be good to be able to adjust the amount of filament pulled in for the purge though.

BastR, 2021-09-16:
@edwardhorsford are you using the firmware 4.3.3 RC and the bondtech extruder?

BastR, 2022-04-30:
This issue is getting old but is still active for me.

I found 2 ways to avoid the engine blocking:

-I use 2 gcode files, 1 for loading the other for unloading in which I just lowered the speed. 100% success rate

or 

-I manually move the X carriage to its Xmax position and use the firmware load/unload. 99% success rate

linx25, 2022-04-30:
I have made a custom firmware

Prusa-Support, 2022-10-20:
Thanks for bringing this to our attention.

I've reason to believe this is a popular mod, but because this problem is not that frequent, there may be a reason specific reason behind it. If so, most likely firmware tweakings are unnecessary.

In my experience as Tech Support, I recall coming across similar issues (as per the video above) with the MINI stock design - unfortunately, I don't have similar cases at the hand as of now as this is not a frequent issue.
As far as I can tell, these kinds of issues are often addressed as hardware or assembly issues, but our official support may not be able to offer adequate support in the case of 3rd party hardware/design/firmware.
In the case the extruder motor is still Prusa-original, you may want to revert to the genuine Prusa design, firmware and settings to see if the problem persists, and let our [Customer Support](https://help.prusa3d.com/article/customer-support_2287) evaluate if the extruder motor works correctly - or simply replace the extruder motor as a test.

The author/seller of the mod though may be able to offer support and possibly help you find the root cause of the issue.

Michele Moramarco
Prusa Research

lejo1512, 2023-09-19:
I

> Thanks for bringing this to our attention.
>
> I've reason to believe this is a popular mod, but because this problem is not that frequent, there may be a reason specific reason behind it. If so, most [...]

It's not frequent, but it happens sometimes for our customers. Seems like replacing the motor work in most cases.
Could be easily solved by reducing the load speed or increasing the current during these moves.

morten-arntun, 2024-01-03:
I've had this exact issue. Updated to 5.1.2 and still a problem. 
Prusa mini with Bondtech extruder. Would be awesome if we could get a setting via gcode or menu for load/unload speed.

morten-arntun, 2024-01-03:
> Thanks for bringing this to our attention.
> 
> I've reason to believe this is a popular mod, but because this problem is not that frequent, there may be a specific reason behind it. If so, most likely firmware tweakings are unnecessary.
> 
> In my experience as Tech Support, I recall coming across similar issues (as per the video above) with the MINI stock design - unfortunately, I don't have similar cases at the hand as of now as this is not a frequent issue. As far as I can tell, these kinds of issues are often addressed as hardware or assembly issues, but our official support may not be able to offer adequate support in the case of 3rd party hardware/design/firmware. In the case the extruder motor is still Prusa-original, you may want to revert to the genuine Prusa design, firmware and settings to see if the problem persists, and let our [Customer Support](https://help.prusa3d.com/article/customer-support_2287) evaluate if the extruder motor works correctly - or simply replace the extruder motor as a test.
> 
> The author/seller of the mod though may be able to offer support and possibly help you find the root cause of the issue.
> 
> Michele Moramarco Prusa Research

I have this exact issue. Updated to 5.1.2 and still there are issues. 
Prusa mini with Bondtech extruder. Would be awesome if we could get a setting via gcode or menu for load/unload speed.

kluvi, 2024-05-25:
Similar problem: https://youtube.com/shorts/7JJ55p3wPew?si=WlZ8WFtuhTc2Z9LM

morten-arntun, 2024-06-24:
> I have made a custom firmware
Is this custom firmware public, and what version is it based on? I would love to get this issue resolved on my printer.

Prusa-Support, 2024-07-13:
> Is this custom firmware public, and what version is it based on? I would love to get this issue resolved on my printer.

@linx25 Thanks for sharing your experience and I believe the last message is meant for you.
As long as you consider your firmware customization safe for use, other users may appreciate testing it 🙂 but this is ultimately up to you. I believe that even the vendor of the mod, or the manufacturer of the parts, may be interested.

Michele Moramarco
Prusa Research

closed 2024-09-19 by github-actions[bot]