prusa3d/Prusa-Firmware-Buddy #1594: [BUG] Selftest fails without steel sheet being placed on the heatbed

Barnabasek, 2021-09-16:
MINI, 4.3.3-RC (Original)

Self-test fails with Heatbed heating error if you run the Selftest without a steel sheet on the heatbed. This bug seems to be present since the first release of the MINI firmware.

This situation of steel sheet not being present on the heatbed during the testing should be either accounted for in the testing algorithm, or a prompt to place the steel sheet on the bed before the Selftest should be added.

milstrata, 2023-07-29:
This issue also exists in the XL semi assembled. The assembly instructions and self-test on the printer don't prompt to put a print sheet on the heated bed, but the heated bed self-test will not pass without a print sheet on it.

I was only able to figure it out and pass self-test because of this post on the forums: https://forum.prusa3d.com/forum/original-prusa-assembly-and-first-prints-troubleshooting/heater-test-fails/

My recommendation is to add a prompt to the self-test on the printer that instructs the user to put a print sheet on the bed.

Prusa-Support, 2023-07-30:
Hello @milstrata, thank you for the feedback on this. I confirm this behaviour is not a bug and due to the fact that not having the sheet put on causes the heating of the bed to be already out of "what the printer expects", hence causing the error. 

Good point about the possible prompt, I relayed feedback to our devs on this aspect. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-17 by github-actions[bot]