prusa3d/Prusa-Firmware-Buddy #1599: [BUG] Preheat function heats up hotend only to 170 degrees

JakubKohoutek, 2021-09-17:
MINI 4.3.3-RC - Original + filament runout sensor, bondtech extruder

When I try to preheat the hotend using the predefined profiles (PETG, PLA,...) from the printer's menu, the bed reaches the temperature fine, but the hotend temperature stops at 170 degrees Celsius (no matter what profile I select) and it doesn't go higher. The temperature indicator on the display is blinking so that it looks like it's still heating up, but it isn't.

I tried to preheat the printer using Octoprint by setting up the temperature to 220 degrees there and that works correctly. The printer reaches the temperature without any issue, so the problem is not probably in the hardware.

murk-sy, 2021-09-18:
Preheat heats to 170°C to prevent nozzle ooze onto the bed. The intent is to make the printer ready to start printing as soon as possible. Having bed pre-heated allows immediate start of automatic bed levelling, and having the nozzle heated but under the actual print temperature helps keep the bed clean until the printer is ready to actually start extruding.

JohnnyDeer, 2021-09-20:
Hi @JakubKohoutek, as @murk-sy wrote this behavior is feature to prevent nozzle ooze onto the bed. Temperature of used material is reachen after mesh bed leveling and if you want to load/unload filament.

JakubKohoutek, 2021-09-20:
Hi @murk-sy and @JohnnyDeer, thank you for the explanation. I understand it's a "feature not bug" kind of behavior, but IMHO it's pretty confusing. In my case, I had a clogged nozzle and I needed to clean it up by heating the hotend as much as possible and do a cold pull, which wasn't possible to do properly when preheating stops at 170 C. I had to use Octoprint for that, but there are people who don't have this option.

murk-sy, 2021-09-20:
Preheat is intended for print preparation since it also heats the bed, though to be very honest I would actually prefer a function to fully heat just the nozzle for cleaning/changing/filament change. I would actually be interested in how many people use it since it *can* save you *maybe* 3 minutes if you work in the correct order, I suppose more for higher temperature materials.

As I've been writing and rewriting this, I've worked out an idea to hopefully make us happy too, and it should be relatively easy to implement. @JohnnyDeer Would like to hear what you think.

Since you generally don't need to preheat for non-loaded materials, I suggest the preheat menu layout is changed like this:
- Preheat for printing PLA (170-215/60)
- Heat nozzle for PLA (215)
- Other materials
  - Preheat PLA
  - Preheat PETG
  - Etc (the standard menu)
- Cooldown

Displaying 170°c in that menu for preheat would be nice but I understand there's limited space. Perhaps current material could be put on top as a heading?

This way,
- Preheat for current material is accessible by just double clicking on preheat
- Cooldown is still accessible by clicking preheat and scrolling down
- An easy way to heat up just the nozzle is added
- The old options are still available

anttikarhu, 2021-09-24:
> Hi @murk-sy and @JohnnyDeer, thank you for the explanation. I understand it's a "feature not bug" kind of behavior, but IMHO it's pretty confusing. In my case, I had a clogged nozzle and I needed to [...]

You can use Settings > Temperature menu to set any nozzle temperature. No need for Octoprint. You do this also when changing the nozzle, doing cold-pull etc, and it’s well documented in the guides.

Pre-heat works as intended - not a bug :)

JakubKohoutek, 2021-09-24:
@anttikarhu OK, thank you for the explanation. I still believe it would be more intuitive to add an option for nozzle preheat (before nozzle change or cold pull) to the preheat menu, but I agree it's not a bug - closing the issue 👍

closed 2021-09-24 by JakubKohoutek