prusa3d/Prusa-Firmware-Buddy #1603: How to update?

fragged3d, 2021-09-17:
What file do I need to flash to the mini?

Gordontgopher, 2021-09-17:
You need a .bbf file. The file for the latest release is at the bottom of this page: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2

JohnnyDeer, 2021-09-20:
Exactly as @Gordontgopher write, thank you. @fragged3d was you attempt to flash printer successful?

fragged3d, 2021-09-20:
Yes, but I’m not seeing the way to reverse the extruder motor and the steps! So I’m still working on it!

Thanks,
Ryan Sherer

Gordontgopher, 2021-09-20:
In that case you need the 4.3.3 Release Candidate https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3-RC there are details there on changing steps etc.

fragged3d, 2021-09-20:
That’s the firmware on it and still no options!

Gordontgopher, 2021-09-20:
Have you read the release notes? There is a section there(Support for custom extruders) which includes the instruction "To adjust settings for your custom extruder, go to LCD Menu -> Settings -> HW settings -> then long-press the knob. Similarly to a MK3S+ a new “Experimental” submenu will appear on the screen, where you can tweak the following:...."

If you are on 4.3.3 RC1 and can't activate the hidden sub menu following these instructions then I'd suggest that you need to close this issue and open a new one specific to the problem you are seeing, so the developers can take a look.

fragged3d, 2021-09-20:
I figured it out thanks!

JohnnyDeer, 2021-09-23:
@fragged3d great to hear, so can you please close this issue?

closed 2021-09-23 by fragged3d