prusa3d/Prusa-Firmware-Buddy #1619: [BUG] ODOMETER: "Print time" is not updated when printing from Pronterface

jcjames13009, 2021-09-28:
**Printer type** - [MINI]

**Printer firmware version** - [4.3.3-RC]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or Pronterface**
  
**Describe the bug**
Checking Odometer feature, "Print time" is not updated when printing from Pronterface 😟 Other data is ok (X Y Z axis and filament)
All data is recorded properly when printing from USB stick 👍 

**How to reproduce**
Launch any print from Pronterface and check that "Print time" is not updated
Launch any print from USB and check that "Print time" is updated

**Expected behavior**
Have "Print time" updated when using Pronterface

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

kohend, 2021-11-25:
Seen a similar case with the i3 firmware.

gimpfenlord, 2021-12-07:
same with octoprint

closed 2024-06-17 by github-actions[bot]