prusa3d/Prusa-Firmware-Buddy #1621: [BUG] 4.3.3-RC Mintemp errors

webRat, 2021-09-30:
**Printer type** - Prusa Mini+
**Printer firmware version** - 4.3.3-RC
**Original or Custom firmware** - Original

**Describe the bug**
I printed PLA material for well over a week with no issues, today, I switched to PETG, and now I'm getting mintemp errors left and right. I downgraded back to 4.3.2 and now everything is normal again.

**How to reproduce**
Just run 2 calibration tests, and then attempt to print any model.

**Expected behavior**
I expected to be able to run as smoothly as I did in 4.3.2

**Crash dump file**
I forgot about this and will check on the drive, I don't know that it's there, but I'll look after I'm done printing.

JohnnyDeer, 2021-10-05:
Hi @webRat, could you please provide more info? Are those errors random or in same part of printing? Can you please upload Gcode? Video with behavior just before watchdog would help also. Thank you

JohnnyDeer, 2021-11-08:
We need more informations, otherwise this issue will be closed soon.

closed 2022-01-17 by JohnnyDeer