prusa3d/Prusa-Firmware-Buddy #1662: [FEATURE REQUEST] Add a "Pause Printing" menu option

rgesswein, 2021-11-01:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - Mini+

**Describe the expected functionality**
  Sometimes I need to pause printing while the printer is connected over serial port (Octoprint), would be nice to have that option in the Tune menu. Similar behavior of Change Filament, but just pause printing.

rgesswein, 2021-11-01:
.

hertzi, 2022-03-04:
@rgesswein  add label 'octoprint' and delete the first three lines - its a bit confusing

closed 2024-06-18 by github-actions[bot]