prusa3d/Prusa-Firmware-Buddy #1668: [FEATURE REQUEST] "Unload Filament after print" option

SnickCH, 2021-11-05:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches: I couldn't find anything with the search that matches this. 

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI+]
**Optional upgrades** - [Filament Runout Sensor - not necessary]

**Is your feature request related to an existing problem? Please describe.**
After a print the Mini+ cools down the nozzle and printbed (as expected). if I now want to unload the filament, I have to use the option "unload filament" manually. Everything is reheated and needs time.  This is a waste of time and energy if I already know from the beginning, that after ther print is finished, I will unoad the filament and store it (and not load another filament).

**Describe the expected functionality**
 I would like to have an option "unload filament after print is finished" after the print is started - could be implemented like the option  to change the Z-offset. When this option is activated, the printer should unload the filament automatically after the print is finished. After the filament is unloaded it should cool down (as it is now). 

This would safe time and energy.

abacabie, 2021-11-05:
I like this idea, not sure if it could be implemented but if it was i can see it been very useful.

sfrayy, 2021-11-06:
Yes please, this would be a great addition!

SnickCH, 2021-11-06:
> I like this idea, not sure if it could be implemented but if it was i can see it been very useful.

I don't see why it couldn't be done. On my Creality printer I do the same with some "endscripts" in simplify3d. There I tell my printer to retract xx cm of filament. Of course then it is not unloaded like it is here, but at least the filament is not anymore in the nozzle and I can just unload it manually after the printer is cooled down. No need to reheat it.

Here (with my feature request) you would put it in the firmare. After every print is finished it will call a function that will start the unloading. If necessary it could ad some end-script commands into prusa slicer that will call this function, but I guess it could be done directly in the firmware (and would make the most sense). In the end it's "just" a software solution, therefore I see no reason why it shouldn't be possible. It should even work without a filament sensour: prusa knows how long the standard PTFE-tube is from the extruder to the nozzle - this is the lenght (maybe + some extra security lenght) that needs to be pushed out after a print. 

I think this would be a quick-win for Prusa and a big help for the users

kohend, 2021-12-03:
You could do it in the slicer end code, but I agree this could be a nice feature.

jh-vo, 2021-12-08:
This would be very helpful, I often watch the end of the print so that I can eject the filament immediately after printing before everything cools down...

SnickCH, 2021-12-08:
> You could do it in the slicer end code, but I agree this could be a nice feature.

Yes, It could be done with a customized end code (I have done it in S3D for my CR10). But why not just ad it as a feature? This can't be hard to implement, would work out of the box and make it a bit easier for everybody.

schleifi, 2021-12-13:
It is usefull to have it in the eg „Tune“-section: you can print several times the same gcode, but only for the last print you activate the „M700“ - without the need to have two files only differing in the „M700“ code.

closed 2024-06-18 by github-actions[bot]