prusa3d/Prusa-Firmware-Buddy #1683: [BUG] Blue screen after the remove filament process

Yohannfra, 2021-11-11:
**Printer type** - Prusa Mini+

**Printer firmware version** - 4.3.3

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
The bug happen with USB and with octoprint

**Describe the bug**
This blue screen appear after the remove filament process (Filament -> Unload filament) finishes.

[image]

**How to reproduce**
  Click Filament -> Unload filament and it sometimes happen.

**Expected behavior**
No blue screen

Thanks you

JohnnyDeer, 2021-11-18:
Hi @Yohannfra, can you please provide more information? How often it happens? It is happening in same part of unloading? Do you have OctoPrint or ethernet plugged in when this happen? With what material it is happening? 
Thank you

Yohannfra, 2021-11-18:
Hello,
It happens sometimes mut not all the time, it's a bit random.
I had the issue with Octoprint connected but also without it.

It happened with ABS and PLA but I didn't use any other material.

It happens at the end of the process and I think and it is always the case, I will try to catch it on vidéo to show you.

Thanks.

JohnnyDeer, 2021-11-18:
@Yohannfra thank you for info. Yes please video would be helpful.

Yohannfra, 2021-11-20:
Hello @JohnnyDeer,
Here is the video from the boot of the prusa mini to the crash after the remove filament process : 

https://www.youtube.com/watch?v=rE6tmUYNync

Thanks you.

JohnnyDeer, 2021-11-22:
@Yohannfra thank you. Unfortunately we are not able to reproduce it. 
Can you please upload also crash dump?

JohnnyDeer, 2021-11-22:
One more thing could help to troubleshoot it, please try to unplug E-motor and then do unload procedure. Thank you

Yohannfra, 2021-11-22:
Hello @JohnnyDeer how can I generate the crash dump ?

Unplug E-motor so it won't remove the filament ?

Thanks

JohnnyDeer, 2021-11-23:
@Yohannfra to download the Crash Dump file, insert a USB drive into the printer first and then head to “Settings -> Save Crash Dump”.
Unplig E-motor to determine if the BSOD is caused by an undervoltage on the motor.
Thank you

mtaumike, 2021-11-23:
This is the same kind of bug that sometimes happens when you are in a sub-menu and reconnect USB serial (like Octoprint) and has been reported before.

It is very random.

DavidB1975, 2021-12-07:
I get the exact same error if i use the unload/load function - if I use change filament I don't think it happened yet.

jpautom, 2021-12-21:
It happens to me ~33% of the times I "change" filament. It happens immediately after the old filament has been unloaded. See crash dump.

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/7757518/dump.zip)

Yith1, 2022-01-14:
I get this happening about 33% of the time, but only during a manual filament change, changes performed during a print with Octoprint work fine for me. I don't print from USB ever.

closed 2024-06-18 by github-actions[bot]