prusa3d/Prusa-Firmware-Buddy #1691: [BUG] Icons go missing after inseting a USB stick

reedy, 2021-11-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.3.3

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB

**Describe the bug**
After plugging in a USB stick, the GUI changed and icons went missing

[image]

**How to reproduce**
Not sure if repeatedly replicable.

**Expected behavior**
Icons not disappear

**G-code**
n/a

**Crash dump file**
n/a

**Video**

https://user-images.githubusercontent.com/67615/141703523-ab586471-82f2-46a3-81a8-0907e34055ad.mov

DRracer, 2021-11-15:
:open_mouth: never seen anything like this before. 
Could you please try another USB drive?
Is it reproducible?
Does the GUI respond to knob clicks in this state - i.e. is the FW alive somehow?

reedy, 2021-11-15:
> Does the GUI respond to knob clicks in this state - i.e. is the FW alive somehow?

Yeah, see the attached video :)

reedy, 2021-11-15:
I've not managed to reproduce it, and as such, not so much value in testing other USB sticks.

For reference, it is the Prusa supplied stick with the Mini.

I'd spent a good 6-8 hours doing variosus prints (of 1-2 hours in length) yesterday. Taking the USB stick out after one print, copying over the gcode from my rMBP, plugging it back in again. Some repeat prints too, using the already saved gcode.

Dunno if it's related, or mostly an anecdote.. My rMBP (early 2015) doesn't seem to like that USB stick too much. It seems very slow to recognise the usb stick, slow to write files to it, and *very* slow to eject. Quite often having to force eject. I think before plugging it in last night and getting the "error"... It might've been a force eject because macOS decided it was taking too long to eject etc

zoltan-l, 2021-12-27:
How did you manage working again?

- Unplug usb
- Reset
- Power recycle

reedy, 2021-12-31:
I turned it off and on again using the power switch

closed 2024-05-17 by reedy