prusa3d/Prusa-Firmware-Buddy #1700: [BUG] "Print will end time" increments by 24 hours when the TZ is modified during a print

tbenz9, 2021-11-19:
**Printer type**
MINI+

**Printer firmware version**
4.3.3

**Original or Custom firmware**
Original

**Optional upgrades**
Filament Runout Sensor

**USB drive or USB/Octoprint**
USB Flash Drive

**Describe the bug**
When the timezone, UTC offset (TZ) is changed during a print the "Print will end time" increments by 24 hours.  Every time you change the TZ the "Print will end time" will increment by another 24 hours.

See the linked video!

**How to reproduce**
1. Set the TZ to `-7`
2. Begin a print
3. Take note of the "Print will end time"
4. Change the TZ to `-8`
5. Take note of the "Print will end time" (it will be 24hrs off)
6. Change the TZ to `-9` 
7. Take note of the "Print will end time" (it will be 48hrs off)
8. Change the TZ to `-8`
9. Take note of the "Print will end time" (it will be 72hrs off)

**Expected behavior**
"Print will end time" would be adjusted to reflect the new TZ and appropriate end time.

**Video**
See video here for a very clear reproducer!  https://photos.app.goo.gl/peK8NUTGADP7iNVYA

zoltan-l, 2022-01-12:
tbenz9
Thanks for the report we are looking into it

zoltan-l, 2022-01-28:
Hi @tbenz9 
Can you please let us know what weekday and day time you made the attached video? We are trying to reproduce tha case, so we need as much as possible details. We suspect, that this is happening when the day time, time zone and print estimated time combination is crossing the "midnight"
Thanks

tbenz9, 2022-01-29:
The video was taken on Thursday, November 18th, at 22:52 in the timezone GMT-07:00.  The print estimated time would have crossed midnight as you mentioned.

I have not attempted to replicate this issue again but I can try tonight if that would be helpful.

zoltan-l, 2022-01-30:
would be great, thanks as I did not succed to replicate

tbenz9, 2022-02-19:
@zoltan-l I am able to reliably replicate this bug if my print estimated time passes "midnight" and I change the timezone.  

In the first video my local time was 22:52.

Here is another video showing the same behavior when I update the TZ from -6 to -8, -9, -10 etc.
https://photos.app.goo.gl/faeqemKDAmm8wTMV8

Prusa-Support, 2022-12-22:
The new firmware [4.4.1-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1-RC1) improves the estimated end time thanks to the adoption of the RTC (Real Time Clock), and possibly solves this issue.
Please mind that the _timezone_ still needs to be _manually set by the user_.

Could this issue be closed?

Michele Moramarco
Prusa Research

tbenz9, 2022-12-22:
Hi Michele, thanks for the response.  My expectation is Prusa should have more confidence in if the RTC change actually fixes the issue or not.  Your response "possibly solves this issue" sounds to me like the fix was never tested or verified.  I'm more than happy to see this issue closed if the fix is tested and verified.  Thanks again!

Prusa-Support, 2023-02-01:
The final release [4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) is out.
This issue doesn't persist and will therefore be closed.

Feel free to try and comment back, feedback is always most appreciated.

Michele Moramarco
Prusa Research

closed 2023-02-01 by Prusa-Support