prusa3d/Prusa-Firmware-Buddy #1702: Improve UI message when loading filament

joneuhauser, 2021-11-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.3.3]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint** - [USB]

**Describe the bug** 

When changing filament, it prompts to "Press CONTINUE and push filament into the extruder." (https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/gui/dialogs/DialogLoadUnload.cpp#L24). However, the filament needs to be inserted (touching the extruder gears) before pressing continue, because pressing continue rotates the extruder gears.

**How to reproduce**
Load filament.

**Expected behavior**
The message should be "Push filament into the extruder and press CONTINUE.".

mtaumike, 2021-11-23:
ha! I never caught this before! Good job!

zoltan-l, 2021-12-17:
It is not really a programming bug, I would see it as guidance enhancement , but a very good one. let us see!

PiorGit, 2022-08-25:
@zoltan-l Well, it isn't so much a mere guidance enhancement since the current instruction state literally the *opposite* of what one should do :D
Just wanted to confirm that this is indeed quite annoying, as I cannot see how a new user could ever guess the proper order of operations.

TheOneTrueRazor, 2022-08-25:
> It is not really a programming bug, I would see it as guidance enhancement , but a very good one. let us see!

Instructing people the right way to do a procedure is an "enhancement"? 🤔 Sorry, @zoltan-l this issue is a lot closer to being a bug than it is an enhancement. Maybe if it were labeled as such, this very simple "fix" would have at least been assigned to someone by now.

PiorGit, 2022-08-25:
@TheOneTrueRazor Unfortunately this kind of miscommunication happens all the time in software development : Users report a UX issue using the only means at their disposal (a bug tracker), an engineer then (rightfully) replies that "this isn't a bug" (because of course it isn't one, since there is no functional code to fix) ... and the issue remains unresolved.

This should probably be flagged as "UI" rather than "enhancement".

Prusa-Support, 2022-08-26:
Hello! Thank you for the comments and sorry for the late response from my side. I labelled the issue as UI related and brought it to our devs attention. 

Alessandro Pantaleo
Prusa Research

PiorGit, 2022-08-26:
That's appreciated ! Much thanks.

TheOneTrueRazor, 2022-08-26:
> @TheOneTrueRazor Unfortunately this kind of miscommunication happens all the time in software development : Users report a UX issue using the only means at their disposal (a bug tracker), an engineer then (rightfully) replies that "this isn't a bug" (because of course it isn't one, since there is no functional code to fix) ... and the issue remains unassigned and unresolved.
> 
> This should probably be flagged as "UI" rather than "enhancement".

"UI" is great suggestion. 👍

Prusa-Support, 2022-09-13:
Hello! Thank you for the additional suggestions, I brought these also to our devs attention. In case of anything in the meantime, both suggestions or anything to review that can be related, do not hesitate to add them with further comments. 

Alessandro Pantaleo
Prusa Research

PiorGit, 2022-09-14:
Hello Alessandro ! Great to hear that it is being considered.

FWIW there is still an ongoing discussion going on on the topic on the public forum - which, as one would expect, lead to some users claiming that the instructions are 100% correct as they are, and that the user should be bright enough to figure out the order :D That's interesting because it shows precisely the kind of thinking that must have lead to the ambiguous instructions in the first place (ie thinking of them like a programmer, as opposed to thinking of them like a UX designer).

https://forum.prusa3d.com/forum/general-discussion-announcements-and-releases/the-instructions-for-filament-change-are-incorrect-as-they-are-presented-in-reverse-order/#post-620219

closed 2024-06-18 by github-actions[bot]