prusa3d/Prusa-Firmware-Buddy #1710: [BUG] 4.3.4-RC uses old bootloader

kohend, 2021-11-25:
**Printer type** - [MINI]

**Printer firmware version** - 4.3.4-RC

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint**
  N/A

**Describe the bug**
  Despite the master is set to use version 1.1.1 bootloader the RC version is still using the old bootstrap file.

**How to reproduce**
  N/A.

**Expected behavior**
  An up to date bootloader version.

DRracer, 2021-11-25:
@kohend I'm afraid your expectation doesn't match the technical reality. 
Bootloader is preloaded on the Buddy board from factory, it cannot be changed. Only firmware can be updated.
So no firmware update can change the bootloader no matter what is the latest bootloader version

kohend, 2021-11-26:
So if that case why are the builds created with a bootloader?

zoltan-l, 2022-01-12:
@kohend 
Are you used to build your own builds? The boot parameter is not defining whether you are going to build with/without bootloader

closed 2024-06-18 by github-actions[bot]