prusa3d/Prusa-Firmware-Buddy #1711: [ENHANCEMENT] Switch Sheets Without OctoPrint Disconnect

chrishilbert, 2021-11-25:
**Printer type** - [MINI]
**Optional upgrades** - Satin Sheet/Textured Sheet/PEI Sheet

**Is your suggested improvement related to an existing problem? Please describe.**
To switch sheet profiles, you must disconnect OctoPrint in order to do so.

**Describe the expected functionality**
Add to the Tune menu an option to switch the sheet profiles without requiring an OctoPrint disconnect/reconnect.

tverett, 2023-01-22:
+1 tune menu is an obvious place for it.

Prusa-Support, 2023-02-23:
Thanks for your suggestion.

Octoprint is a third-party software. Several options may not be compatible as of now.
Our developers are following tight and intense development plans at the moment but they should get their chance to look into this in the future.

Michele Moramarco
Prusa Research

closed 2024-06-18 by github-actions[bot]