prusa3d/Prusa-Firmware-Buddy #1714: How to prepare Prusa Mini WiFi module

Erik262, 2021-11-26:
Hello there, 

I want to help and troubleshoot with the WiFi module for the Prusa Mini. Is there any instruction on how to flash an ESP8266 (ESP01) to get this working on the Mini? 

I know there is a folder called "ESP" with some files in there. Since I'm familiar with programming, you don't have to write a complete instruction, just a few hints so that I can go through it.

I really appreciate any help you can provide.

x10102, 2021-11-27:
AFAIK the WiFi module support isn't there yet and will be added in a future update.

Erik262, 2021-11-27:
It was mentioned in the last Prusa Livestream that we could try the new WiFi (module) by checking out the WiFi branch, which is already merged with the master branch. And in the Formnext Exhibition, it was mentioned that the Prusa XL has WiFi.

crasu, 2021-11-28:
Probaly one of the at firmwares here I guess https://www.espressif.com/en/support/download/at

Erik262, 2021-11-28:
> Probaly one of the AT-firmwares here I guess https://www.espressif.com/en/support/download/at

I thought the ESP has to be flashed with a special firmware provided by Prusa?

crasu, 2021-11-28:
They seem to use the standard client lib for the AT firmware:
https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/lib/ESP/lwesp

Have you tried it? I do not have an esp01s to try at the moment.

Erik262, 2021-11-29:
> They seem to use the standard client lib for the AT firmware: https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/lib/ESP/lwesp
> 
> Have you tried it? I do not have an esp01s to try at the moment.

Ahhh you are right. I will try this out when my ESP has arrived.

RyanDam, 2021-12-04:
I spoted a PR which describe how to use esp01 with Prusa Mini here: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1743

Currently I'm not have any esp01 in my hand, so can anyone test it?

Erik262, 2021-12-04:
> I spoted a PR which describe how to use esp01 with Prusa Mini here: #1743
> 
> Currently I'm not have any esp01 in my hand, so can anyone test it?

You are right! Here is the full instruction on how to get the Prusa Mini working with WiFi: 
https://github.com/zoltan-l/Prusa-Firmware-Buddy/tree/ESP-How-to/doc/ESP

foobarjimmy, 2021-12-07:
> > I spoted a PR which describe how to use esp01 with Prusa Mini here: #1743
> > Currently I'm not have any esp01 in my hand, so can anyone test it?
> 
> You are right! Here is the full instruction on how to get the Prusa Mini working with WiFi: https://github.com/zoltan-l/Prusa-Firmware-Buddy/tree/ESP-How-to/doc/ESP Will try this when my ESP01 arrived.

I've compiled latest buddy firmware from master branch and plugged in a esp01s.It worked for the wireless connection but froze forever once I return to the main menu.So if I don't go back to main menu everything works,trying to connect prusalink would result in a Connection Reset.

foobarjimmy, 2021-12-07:
It does have esp support now but I'd say it's still too early to be stable.

Erik262, 2021-12-07:
> It does have esp support now, but I'd say it's still too early to be stable.

Have you tried the latest official version? Maybe they did break something on the master branch and didn't fix it yet.
(I don't know whether there is any ESP support for 4.3.3)

foobarjimmy, 2021-12-07:
> 

esp update menu item(in settings) is not present in both official firmware 4.3.3&4.3.4.So users can't flash from the printer yet.The only way to access the menu item is to compile latest firmware.

Erik262, 2021-12-07:
Since you already flashed your ESP, shouldn't it still work for the official 4.3.3? The ESP could also be flashed with a serial UART thing.

foobarjimmy, 2021-12-07:
> 

I didn’t try flashed esp module on 4.3.3.I’ll give it a try.

foobarjimmy, 2021-12-08:
esp module works on 4.3.3 latest commit.I followed this guide. #1751

RyanDam, 2021-12-08:
> esp module works on 4.3.3 latest commit.I followed this guide. #1751

Great new! May I ask which version of Prusa Link is it? Is it support printer control (move x, y, z from web) or upload gcode to printer?

foobarjimmy, 2021-12-08:
> > esp module works on 4.3.3 latest commit.I followed this guide. #1751
> 
> Great new! May I ask which version of Prusa Link is it? Is it support printer control (move x, y, z from web) or upload gcode to printer?

Still pretty old version.But I've spotted a new prusa link by vorner #1722,that branch currently fails to generate a bbf firmware and has failed CI test.

No control now I'm afraid to say.Just a web service monitoring mini printer.Howerver it's possible to use prusalink as a json source for other projects that involve printer status monitoring.

RyanDam, 2021-12-08:
> Still pretty old version.But I've spotted a new prusa link by vorner #1722,that branch currently fails to generate a bbf firmware and has failed CI test.
>
> No control now I'm afraid to say.Just a web [...]

I'm looking forward for that PR. I watched the XL announcement video and spotted that they show off a new Prusa Link webpage with much more features. I hope this is the case so I can free my Pi4 :D

foobarjimmy, 2021-12-08:
> I'm looking forward for that PR. I watched the XL announcement video and spotted that they show off a new Prusa Link webpage with much more features. I hope this is the case so I can free my Pi4 :D

That feature in the video looks more like a cloud printing solution with local hardware.Would that be something completely different from prusa link local?

Erik262, 2021-12-08:
I'm pretty sure this is for the newly announced Prusa PRO AFS [YouTube Link](https://youtu.be/uLMRAC2zJJA?t=61). But why it shouldn't work for the mini? Well, we have to wait.

Mryck, 2021-12-12:
Hi all I manage to get the wifi and prusaLink  working, the process to flash the ESP is not to difficult but as it's still in development you will have some bug, for example I get a lot of 'Cannot connect to printer' on PrusaLink and after an upload with the checkbox to start the print once the upload is done I have a total freeze and a BSOD. Some screenshot of the New dashboard:
[image]
[image]

For devs here is the BSOD that I face:

[image]

Emanuelgtxf, 2021-12-13:
Muito bom trabalho força. 
Ass. Tec Borges

Erik262, 2021-12-13:
@Mryck did you flash with the prusa or with a serial USB?

Mryck, 2021-12-13:
@Erik262 I flashed the ESP directly with the Prusa with the ESP folder in the USB Stick

Emanuelgtxf, 2021-12-13:
Como?

zoltan-l, 2021-12-14:
Hi guys, find please the guidlines here[ https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/doc/ESP](url)

Mryck, 2021-12-14:
Hi @zoltan-l i think your link is broken. 
On my side I followed the readMe in ./doc/ESP

Erik262, 2021-12-14:
@zoltan-l Could you please provide some instruction on how to flash the esp without using the PrusaMini? Well I know how to do it, but don't know which binary files I have to use.

hertzi, 2022-01-31:
@Erik262 download from [Expressif](https://www.espressif.com/en/products/socs/esp8266ex/resources) the flasher tool and follow the instructions in the pdf

Erik262, 2022-02-01:
very nice @hertzi will try it later. This WIFI thing takes too much time. They should better work more on that feature, since they promised 2 years ago and manufactured a completely new printer within this time during chip shortage and pandemic. lol.

zoltan-l, 2022-02-01:
Hi @Erik262 
this link should work https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/doc/ESP

zoltan-l, 2022-02-01:
Hi @Erik262 will this satisfy?

#define BOOT_ADDRESS      0x00000ul
#define USER_ADDRESS      0x01000ul
#define BLANK1_ADDRESS    0x7e000ul
#define BLANK2_ADDRESS    0xfb000ul
#define INIT_DATA_ADDRESS 0xfc000ul
#define BLANK3_ADDRESS    0xfe000ul

anyway all addresses and files are here : https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/gui/screen_menu_esp_update.cpp

lines: 31-36 
and filenames lines: 125 - 130

Spacekoaster, 2022-05-16:
Hello, I'm a little late to the party, but I also wanted to try this by myself. I've followed the guide from this page [(https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/doc/ESP)] and I tried to flash the latest firmware avalaible by following the instructions on this page [https://github.com/prusa3d/Prusa-Firmware-Buddy](url) and using the `./utils/build.py --generate-bbf` command. When trying to flash the printer from the USB sticks, it detects the .bbf firmware file, but tells me that it can't verify the authenticity of the firmware and therefore can't flash it because the printer thinks that it is an unofficial firmware. Does anybody knows what I've been doing wrong? Thanks

hertzi, 2022-05-16:
> Hello, I'm a little late to the party, but I also wanted to try this by myself. I've followed the guide from this page [(https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/master/doc/ESP)](url) and I tried to flash the latest firmware avalaible by following the instructions on this page [https://github.com/prusa3d/Prusa-Firmware-Buddy](url) and using the `./utils/build.py --generate-bbf` command. When trying to flash the printer from the USB sticks, it detects the .bbf firmware file, but tells me that it can't verify the authenticity of the firmware and therefore can't flash it because the printer thinks that it is an unofficial firmware. Does anybody knows what I've been doing wrong? Thanks

Hi Spacekoaster, if you compile the firmware yourself it is an unofficial firmware. The official is published and signed from Prusa.
If you want to run your build you have to [break the appendix of the Buddy board](https://help.prusa3d.com/article/flashing-custom-firmware-mini_14)

Spacekoaster, 2022-05-16:
Thank you very much for your answer. Can you confirm me that the official releases actually don't have the "flash ESP" functionality even if the good files are on the USB sticks?

Erik262, 2022-05-17:
> 

They didn't release it yet. But I assume they will be soon an update.

Prusa-Support, 2022-07-31:
Hello! Sorry for the late response to this thread - during these days, we released a new Beta firmware (4.4.0) with support for WiFi. All the details for the release here: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases. However, please note this is a beta release dedicated to experienced users. Some features in this firmware are still being developed and might not behave as expected. Use GitHub and forums to report any issues. 

Alessandro Pantaleo
Prusa Research

Emanuelgtxf, 2022-08-04:
o meu ja da ate agora sem erros

Prusa-Support, 2022-08-08:
Hello @Emanuelgtxf, thank you for your comment. I translated it with Google and I read it's going without errors, so all good with it? In case of issues of course do not hesitate to elaborate on these, we'll review them in more detail. Thank you! 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2023-01-12:
Hi everyone. It seems that the final release for ESP Wi-Fi module compatibility went well, but this is just the beginning. 🙂 In fact our developers are already polishing a new release.

Please allow me to close this issue.
However, feel free to share feedback on this or any other relevant issue.

Michele Moramarco
Prusa Reserch

closed 2023-01-12 by Prusa-Support