prusa3d/Prusa-Firmware-Buddy #1720: [BUG] Fan test after XYZ axis test triggers reset

NateCrawford, 2021-11-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
MINI+

**Printer firmware version** - [e.g. 4.0.5, ...]
4.3.3

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
Filament Sensor

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint
USB

**Describe the bug**
  A clear and concise description of what the bug is.
After running an XYZ-axis test, started a Fan test. Fans did not spin up, and a watchdog reset occurred. After the reboot, a Fan test succeeds, but doing an XYZ then Fan test (again, no fan spinning) triggered something like a factory reset, as all saved configuration (and odometry) was lost. After this second reset, the XYZ/Fan test combination succeeds. 

**How to reproduce**
  Please describe steps to reproduce the behavior.
Can not reproduce after factory reset

**Expected behavior**
  A clear and concise description of what you expected to happen.
A self-test component should not trigger a reset or loss of configuration.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.
A dump.bin from 1-Jan-1980 exists on the USB that was attached, but it may not be related

**Video**
  Please attach a video. It usually helps to solve the problem.

NateCrawford, 2021-11-29:
Dump file:
[dump.bin.gz](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/7620648/dump.bin.gz)

JohnnyDeer, 2021-12-02:
Hi @NateCrawford, thank you for reporting. Unfortunately we are not able to reproduce this behavior. Do you remember what actions you performed on the printer before faulty selftest?

NateCrawford, 2021-12-08:
I am not able to reproduce, either. This is what I can remember doing:
1) Successfully print an 8-hour part, but with a little noise on the Z-axis when done.
2) Unload filament, let system cool to room temp.
3) Manually drive axes (through menu, not actual hands).
4) Add a little lubrication on z-axis screw where it was rubbing a bit, tightened x-axis belt.
5) Ran XYZ axis test several times until things sounded fine.
6) Ran the fan test (for no real reason other than it was right there) --> fans did not spin up, watchdog reset.
7) After reset, ran XYZ test --> OK
8) Ran fan test --> fans did not spin, apparent factory reset (no message on display)
9) Lost all settings, but XYZ and fan tests work properly now.

If someone else hits this in the future, hopefully they can give better data. Since the Mini appears to be working fine now, I'm good with closing this.

Thanks!

SnickCH, 2021-12-20:
I had the exact same issue yesterday during the FAN test

JohnnyDeer, 2021-12-20:
@SnickCH thank you for describing steps before bug appear, I'll try to reproduce it.

GithubUser99999999, 2022-12-16:
I experienced the same bug for the first time now after I changed the nozzle to 0.6 mm and therefore had to run the self test again since months. It constantly gave me a crash plus watchdog error until I removed the USB drive and then it worked. Firmware is 4.4.0

Prusa-Support, 2023-01-17:
Thanks for reporting.

Would the issue be definitely solved after a factory reset?
Would you be able to reproduce the issue. and generate a dump file (_Settings > Save Crash Dump_) after?

Michele Moramarco
Prusa Research

closed 2024-06-18 by github-actions[bot]