prusa3d/Prusa-Firmware-Buddy #1753: Blue screen

fragged3d, 2021-12-06:
My mini keeps blue screening and my print fan doesn’t come on when it’s supposed to? Prusa even sent me a new fan and board! Please advise!

JohnnyDeer, 2021-12-06:
Hi @fragged3d, for begin we need more information. When BSOD appears? During selftest as in #1720 or while printing? Are you using OctoPrint or USB even extension cable can have influence...? Can you upload crash dump file? If it happens during print please uplad Gcode too.
Thank you.

fragged3d, 2021-12-06:
@JohnnyDeer my bsod happens when I reboot but it’s very random! I don’t use octopring and I don’t have an extension cable for my usb!

fragged3d, 2021-12-06:
My print fan not spinning is a constant!

fragged3d, 2021-12-06:
@JohnnyDeer I tried flashing back to prusa stock firmware and it showed the printer flashing it and shows 4.3.3 but it’s still bondtech firmware?!

JohnnyDeer, 2021-12-06:
So you are using Bondtech upgrade, right? Did you ensured that Bondtech upgrade is installed right?  
Sometimes when you reboot printer then BSOD appears? Your print fan is not spinning during print, selftest nor when you turn it on in settings menu? 
Are you using original firmware or custom?

fragged3d, 2021-12-07:
@JohnnyDeer yes it’s installed correctly, yes when I reboot the printer it randomly bsod, selftest says ok on the print fan but it won’t spin even when set to manually 100%, I’m using the firmware in this release!

JohnnyDeer, 2021-12-08:
@fragged3d I´m not sure if this issue is related to the FW or HW, would need to troubleshoot with our Tesch support, please contact us through info@prusa3d.com, in the email refer this issue. Thank you

zoltan-l, 2021-12-31:
if it is stiil an issue, can you please describe it in more details, best would be to look to the issue starting page template and follow it.

Prusa-Support, 2022-05-17:
Hello! Can you please update us on the situation, and send us a picture of the BSOD, if still actual? I noticed that a few days after starting the thread, we released 4.3.4 FW, have you attempted with it and did it improve the situation? In case, feel free to reach me directly via email info@prusa3d.com mentioning me in the subject ("attn: Alessandro", we'll look into it. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-18 by github-actions[bot]