prusa3d/Prusa-Firmware-Buddy #1758: [ENHANCEMENT] <Enter comprehensive title>

gimpfenlord, 2021-12-07:
Printer type - [MINI]

Is your suggested improvement related to an existing problem? Please describe.
To change filament, you must disconnect OctoPrint in order to do so.

Describe the expected functionality
Add to the Tune menu an option to change filament without requiring an OctoPrint disconnect/reconnect.

imatasic, 2021-12-08:
Do you know you can run M600 in octoprint console, while the hotend is sufficiently hot?
using the custom control editor plugin, you can even create a button for it on the control interface.

gimpfenlord, 2021-12-08:
thx for this
i just found this plugin https://plugins.octoprint.org/plugins/simple_filament_change_buttons/
exactly what i was searching for :)

but having that option also on the printer display menu would be cool too.

Pitel, 2022-01-01:
> Describe the expected functionality
> Add to the Tune menu an option to change filament without requiring an OctoPrint disconnect/reconnect.

It is there, in the Tune menu. But it's grayed out. Why?

coaxial, 2022-02-22:
> it's grayed out. Why?

I wonder too and seems to make swapping filaments while on serial unnecessarily complicated.

closed 2024-06-18 by github-actions[bot]