prusa3d/Prusa-Firmware-Buddy #1767: [FEATURE REQUEST] PID auto tuning in the menue for Prusa mini/mini+

SnickCH, 2021-12-09:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI, MINI+]

**Is your feature request related to an existing problem? Please describe.**
A lot of people use silicon socks on the hotend, change the original heatbreak from Prusa with one from Bondtech, nozzle change made from different material (like nozzle-x, hardened steel etc.). Therefore the "stock" settings do not match.

**Describe the expected functionality**
In the menue add a point for "auto tune PID", where we can run this 1:1 like it is available on the MK2/3 printer. It should run on a set temperature and then be safed. Feature from here 1:1 https://help.prusa3d.com/ja/article/pid-tuning_2265

sym0nd0, 2021-12-29:
Whilst I agree with the request, as it would be beneficial to have PID tuning as a menu option, why aren't you storing the PID values to EEPROM when tuning through OctoPrint?

```M106 S255; Set Part Cooling Fan to 100%
M303 C8 E0 S215 U1; Cycle the Hotend to 215C 8 times and save the value to RAM
M500; Save to EEPROM
M303 C8 E-2 S60 U1; Cycle the Bed to 60C 8 times and save the value to RAM
M500; Save to EEPROM
M107; Turn off fans
```

You can  see the PID values in terminal, shown after the 8th cycle for each tool. You can then check the values stored in the EEPROM after a restart of the printer
```
M503; View current PID values
```

sym0nd0, 2021-12-29:
See 

> Thermal regulation for custom extruders

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3

But, I do agree that PID tuning and M500 should be made available on the Mini/+, with an onboard settings option too.

malibata, 2022-02-17:
> See
> 
> > Thermal regulation for custom extruders
> 
> https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3
> 
> But, I do agree that PID tuning and M500 should be made available on the Mini/+, with an onboard settings option too.

The post explains why Prusa use their own instead of Marlin algorithm for PID and has nothing to do with PID tuning. PID tuning is necessary feature even for cases where one adds a silicone sock and should be available. Startup G code is not used by printer unless you are printing, something that you have sliced, for example when you are loading the filament or doing calibration which are all firmware functions or if you are printing a file from Prusa site. So IMHO the best option would be to do it in menu same as with MK2x/3x ...

sym0nd0, 2022-02-17:
> The post explains why Prusa use their own instead of Marlin algorithm for PID and has nothing to do with PID tuning.

Taken directly from the linked post

> The results below show clearly that the current algorithm (**a feed-forward driving + PID tuning of the steady state**) is capable of correct thermal regulation in all tested scenarios including the sudden start of a cooling fan.

malibata, 2022-02-17:
> Taken directly from the linked post

Also directly from the post is the subject of that post:
> Due to the support of custom extruders, we received requests from the community to enable Marlin PID in the firmware.
And the subject of this request is:
> PID auto tuning in the menue for Prusa mini/mini+
The excerpt you are referring to confirms that Prusa uses their own algorithm that also uses  PID tuning, obviously, but does not say anything about saving PID auto tuning values which this request is about and therefore has nothing to do with it.

grosiln, 2022-05-13:
Přidaní funkce ladění PID pro hotend do firmware by vyřešila spoustu problému, pokud použijete nerezovou kalenou či titanovou trysku není tiskárna schopna udržet bez kolísaní teplotu na hotendu, ano jde to řešit přidáním Gcode do sliceru, ale je to krajně nepohodlné, nechápu, když tuto funkci má MK3S, proč by tato funkce nemohla být i na MINI, dalším důvodem je změna  heatbreaku nebo celého hotendu, který řeší Vámi neodladěné problémy v originál heatbreaku, kdy dojde k poškození PTFE trubičky zejména při tisku materiálů kolem teploty 250 stupňů, jako ASA,ABS,CEFPETj a jiné materiály náročné na teplotu, pak místo tisku, stále řešíte servis na hotendu. Myslím, že přidáním této funkce do firmware, bude oceněna komunitou MINI jak jedna z velmi očekávaných funkci, zejména po vydaní hotendu REVO od E3DV6, který pokud Vím je dlouhodobým dodavatelem hotendu pro PRŮŠA TISKÁRNY. Prosím tedy o velmi důkladné zvážení přidání této funkce. Děkuji také za Vaší dosavadní práci na firmware.

Prusa-Support, 2022-09-01:
Thank you for your suggestions.

I may seem to understand that the initial request was to add an option for autotuning the PID directly from the printer menu.
This is unnecessary due to the specific Prusa temperatures control implemented in the firmware (which, as per the release notes mentioned above, proved to be effective even in the case of custom setups). As a consequence, it is not possible to store a custom PID value in the firmware.

Because the latest request in this issue is to make the firmware compatible with PID custom values, that request would be considered a duplicate of issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1341.

I hope you don't mind me closing this issue now.

Michele Moramarco
Prusa Resaerch

closed 2022-09-01 by Prusa-Support

SnickCH, 2022-09-01:
@Prusa-Support 

I made the test with different hotends and with a new 50W heater cartridge. It makes a difference if you tune PID (with octoprint) or just let the standard as it is. The difference: with no tuning it needs way to long until the heat is set. At the beginning I have a long period where the heat is overshooting for 10-15degrees. To handle this, I need 4-10x skirt until the heat is down to the point it should be. The same goes for filament load / unload.

Therefore I guess it is dependent to PID tuning.

Prusa-Support, 2022-09-07:
It is not possible to set a PID value on the MINI firmware as of now but this issue is only closed because it is a duplicate.

Please check issue #1341 and feel free to take part to the existing discussion.

Michele Moramarco
Prusa Research