prusa3d/Prusa-Firmware-Buddy #1779: [ENHANCEMENT] Addition of timezones not a full hour from UTC

Gabriel-06, 2021-12-17:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MINI(+)
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**

Currently, in the timezone settings, it is impossible to specify a timezone that is not a full hour offset from the UTC time (e.g. +9:30). As I live in Adelaide and our timezone is +9:30 or +10:30 depending on whether daylight savings is active, the time is always wrong for me.

**Describe the expected functionality**

It would be great to be able to specify timezones that are not a full hour from UTC, either in the form of hours and minutes (9:30) or in just hours (9.5)

I have looked at the code, but I am way too inexperienced with C to implement it myself. I was hoping it was as easy as changing the 1 in `const SpinConfigInt SpinCnf::timezone_range = { { -12, 14, 1 } };` (in src/guiapi/src/menu_spin_config_basic.cpp)  to a 0.5, but as the function is built for working with integers it cannot accept the float value of 0.5. I'd love to help out and make the changes myself but I don't think I have the skills for it sadly.

Prusa-Support, 2023-01-28:
Thanks for the suggestion.
Recently we have been reworking the clock features so it may be a good chance for me to remind our developers about this.

Michele Moramarco
Prusa Research

Gabriel-06, 2023-03-01:
Any updates on this @Prusa-Support?

Prusa-Support, 2023-03-07:
I'm sorry, recently the clock features have been improved in various ways but the timezones haven't been updated yet,
For the time being this issue will remain open.

Michele Moramarco
Prusa Research

dbr, 2023-11-27:
This would be nice - I'm also in South Australia and the print ETA's are wrong because of this.

Note there's also some +45min timezones, most notably in Nepal https://en.wikipedia.org/wiki/UTC%2B05:45 (and [also some small parts of Australia](https://en.wikipedia.org/wiki/UTC%2B08:45), which seems to love obscure timezone offsets)

The suggestion in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/639#issuecomment-657608143 of selecting by region-name may solve this better, along with also possibly solving the daylight-saving offsets

Prusa-Support, 2024-04-04:
This issue is solved by FW [6.0.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC1), which is a Release Candidate and therefore considered to be a rather stable pre-release.

Feel free to give it a go and share further feedback.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-04-04:
P.S. The MINI-specific firmware version is expected to be released in short.

Michele Moramarco
Prusa Research