prusa3d/Prusa-Firmware-Buddy #1783: [ENHANCEMENT] more then -2.0 offset for the Z axis

SnickCH, 2021-12-19:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI/MINI+]
**Optional upgrades** - [e.g. doesn't matter]

**Is your suggested improvement related to an existing problem? Please describe.**
I reassembled my hotend with a new PTFE tube and had to do the first level calibration. I can set -2.0 Z offset during the calibration, but it is not enough (maybe -2.1 or -2.2 would be perfect. Unfortunately there is a limit on 2.0. 

**Describe the expected functionality**
I would expect that the offset is not limited to -2.0 on the Z axis.

philippbeckmann, 2021-12-20:
For this big offset you should readjust the height of your probe: 

https://help.prusa3d.com/en/guide/height-adjustment-of-m-i-n-d-a-superpinda-sensor-mini-_154547

SnickCH, 2021-12-20:
@philippbeckmann Thank you for your input. I'll try that tomorrow.

SnickCH, 2021-12-26:
After a new calibration of the pinda, it was possible with only -1.x (I always had -1.x). Therefore if this is enough, it can be closed.

closed 2021-12-26 by SnickCH