prusa3d/Prusa-Firmware-Buddy #1787: [BUG] Prusa Connect doesn't show real time of finished print, when speeding up printing

Gyrxiur, 2021-12-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.3.4]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive**

**Describe the bug**
  When you speed up printing, finish time on printer LCD updates accordingly, but in Prusa Connect it still shows finish time as it was on 100% speed.

**How to reproduce**
  Print something, speed up print speed and compare finish time on LCD of printer and in Prusa Connect. They won't match.

**Expected behavior**
  Prusa Connect should display same time as printer - adjusted to print speed.

zoltan-l, 2021-12-31:
Hi, can you please let us know what Prusa Connect configuration do you use?

Gyrxiur, 2022-01-01:
I am connected through my router and have latest official firmware. https://www.youtube.com/watch?v=gAKaJnEKuCQ - I've made video of this bug.

zoltan-l, 2022-01-12:
Gyrxiur,
Thanks for additional information we will look into it.

zoltan-l, 2022-01-20:
validated - analysing the issue

Gyrxiur, 2024-05-22:
> This issue has been flagged as stale because it has been open for 60 days with no activity. The issue will be closed in 7 days unless someone removes the "stale" label or adds a comment.

still not fixed

closed 2024-07-30 by github-actions[bot]