prusa3d/Prusa-Firmware-Buddy #1791: [ENHANCEMENT] Move heater block to neutral x position during filament loading to prevent PTFE tube damage

manolitto, 2021-12-30:
**Printer type** - Prusa MINI+

**Issue**
This week I had a surprising incident: during loading of some PLA the sharp tip of the filament (as it is officially recommended in the handbook to cut it strand) pierced through the "extruder to heat block" PTFE tube. See attached image.

[image]

**Suggested Improvement**
Most times during loading the print head is at the right most position. For instance after doing an auto home first or when a print has just finished. Well, in that position the tube forms a narrow curve. My gut feeling tells me that the damage would not have happened when the print head was in a position where the tube is less bended.
So, my solution is to **automatically move** the **print head** to the **left** **prior** to actually **loading** the filament. I guess an absolute x-position of around 20 to 25 would give a nice slightly bended tube which better pipes the filament during loading.

JohnnyDeer, 2022-01-03:
Hi @manolitto, thank you for reporting, this issue is related to #311 and we will make this feature in one of next releases.

TerTerro, 2022-01-29:
Agree on this, same happened
PTFE tube seems very constrained.
[image of damage](https://imgur.com/TuPLbcT)

MistrKernnunos, 2022-03-07:
Implemented in this PR. 
https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1847

Prusa-Support, 2023-01-21:
Thanks for the information.
The issue will be closed now.

Michele Moramarco
Prusa Research

closed 2023-01-21 by Prusa-Support