prusa3d/Prusa-Firmware-Buddy #1792: [BUG] Display/Screen corruption when printing

elandesign, 2021-12-30:
**Printer type** - MINI

**Printer firmware version** - 4.3.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
Set a print running and returned halfway to a corrupt screen (mostly white/inverted, icons rotated - see attachments). Print finished successfully, but the display remained corrupt until I power cycled the machine.

**How to reproduce**
Not certain - have attached the GCode file which was printing in case that was the cause.

**Expected behavior**
A display which doesn't become corrupt...

**G-code**
[Stomp Box.gcode.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/7793807/Stomp.Box.gcode.txt)

**Crash dump file**
N/A - the machine didn't actually crash

**Video & Photo**
https://user-images.githubusercontent.com/21380/147771201-a37a0774-1edb-4371-b31b-2b22286e3f78.MOV
https://user-images.githubusercontent.com/21380/147771219-b71350ff-02c5-443c-a9c0-ecdd773b13ab.JPG

Gordontgopher, 2021-12-30:
This looks like a duplicate of #1200 there are some suggestions there that might help.

JohnnyDeer, 2022-01-03:
Hi @elandesign, as @Gordontgopher wrote this issue is duplicate. I´m  closing this one.

closed 2022-01-03 by JohnnyDeer