prusa3d/Prusa-Firmware-Buddy #1793: [BUG] "Bed leveling failed" during print

murk-sy, 2021-12-31:
**Printer type**
MINI+

**Printer firmware version**
4.3.4-RC

**Optional upgrades**
Filament Runout Sensor

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
At a reproducible point during a print (but not *always*), the printer appears to attempt bed leveling when 8.2 mm of the print are completed or at least partially extruded.

**Notes**
Actual filament used is PETG. See #1400 
This printer has had its motherboard replaced in february due to network related issues

**How to reproduce**
Run the attached gcode file (2/5 chance of it happening?)

**Expected behavior**
Consistent failure or successful print based on gcode

**G-code**
[2021-12-31 bed leveling failed gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/7795973/2021-12-31.bed.leveling.failed.gcode.zip)
Gcode file pulled from USB drive

**Crash dump file**
[2021-12-31 bed leveling failed dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/7795979/2021-12-31.bed.leveling.failed.dump.zip)
I was unable to see any print message history

**Pictures**
Pictures from 2 different instances of this happening:
[image]
[image]
[image]
[image]

JohnnyDeer, 2022-01-04:
Hi @murk-sy, thank you for reporting. This may be caused by wrong reading from USB. Can you please try print this Gcode with another USB stick?
And thank you for uploading Gcode from USB, this is the correct procedure 👍
Info to DUMP is saving only if printer crashed, so there is nothing related to this bug.

murk-sy, 2022-01-04:
I have actually sliced this for 2 printers one after another so that should be fairly easy to do. On the other printer the gcode worked flawlessly. I did the safe removal procedure for both so theoretically it shouldn't have problems.

I will try to switch (so it's printer 2 + stick 3 and printer 3 + stick 2, problem appeared on printer 2 + stick 2) and see what happens. Will report tomorrow after I start the new batch.

@JohnnyDeer The DUMP info was manually saved after the print failed, so there may be something useful.

murk-sy, 2022-01-06:
Unfortunately I haven't had any luck (or misfortune) reproducing this, even switching back to how it failed originally. The most confusing thing is still that it happened twice identically, but not consecutively.

Could this type of issue theoretically be avoided with a line checksum?

Not exactly optimal to put it in every gcode file, but if it's supported when reading from file and prusaslicer could output it, I would probably just leave it on for reliability and to detect when USB storage starts having issues.

Bed leveling could also be blocked from running during a print since that's just kinda a weird thing to do on a mini (and also somewhat dangerous to the printer), but I'm sure there's *some* use case that would require it.

Feel free to close the issue, though it would be great if you consider the above for potential implementation.

closed 2022-01-07 by murk-sy