prusa3d/Prusa-Firmware-Buddy #1795: [ENHANCEMENT] Improve Preheat menu usability

murk-sy, 2022-01-04:
Since you generally don't need to preheat for non-loaded materials, I propose the preheat menu layout is changed to this or similar:
- Preheat for printing PLA (170-215/60)
- Heat nozzle for PLA (215)
- Heat bed for PLA (60)
- Cooldown

Preheat is intended for printing preparation since it also heats the bed, though to be very honest I would actually prefer to have a function to fully heat just the nozzle for cleaning/nozzle change/filament change (which shouldn't heat the bed either in my opinion, but that's just my particular process).

Using preheat to clean the nozzle without accidentally touching the bed and burning my hand would be a lot easier with an extra option like that. Currently, the only way to achieve nozzle-only heating is to remember the temperature required (woe is me, I know) and then go manually change it. It's just a bit of a drag when it could literally be just 2 clicks and make maintenance a lot easier.

Displaying 170°c in that menu for preheat would be nice since that's what it *actually* heats to, but I understand there's limited space. 

With the above suggestion,
- Preheat for current material is accessible by just double clicking on preheat
- Cooldown is still accessible by clicking preheat and scrolling down
- An easy way to heat up just the nozzle is added
- Since you don't *really* need to preheat for non-loaded material, other filament options should be removed. Once filament is changed, the printer is preheated for that material automatically.
 
If you use pre-heat for non-loaded material, please post about it, because it's not something I've encountered before.
(Except as workaround for bug #1400 )

Since this firmware will be/is similar for Prusa XL, doing it this way would allow quality of life improvement for preheating those:
- Tool 1 (PLA)
  - Preheat for printing PLA (170-215/60)
  - Heat nozzle for PLA (215)
  - Auto-clean tool? (assuming that is an issue and not in the filament/tool menu)
  - Other materials
    - Preheat PLA
    - Preheat PETG
    - Etc (the standard menu)
  - Cooldown
- Tool 2 (PETG)
  - Preheat for printing PETG (170-230/80)
  - Heat nozzle for PETG (230)
  - Auto-clean tool?
  - Other materials
    - Preheat PLA
    - Preheat PETG
    - Etc (the standard menu)
  - Cooldown
- Bed
  - Bed specific options (quadrant/all heating, will likely be too complicated to efficiently control directly except for troubleshooting)
- Cooldown All

I wrote this as a comment on a different issue that was closed, so after I remembered it again I decided to post this separately as an enhancement.

Prusa-Support, 2022-09-27:
Thank you for your suggestion. 

This issue extends the existing issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/706, or actually partially duplicates it.
Because this is a good piece of advice, I would like to display a mention in the main issue without closing this one.
The final decision about how to implement the feature, and the user interface, will be up to the developer's team, but thank you again for the suggestion.

Michele Moramarco
Prusa Research

Eldenroot, 2022-11-26:
Would be nice, because it could extend the life of the heater and other components. In many cases not necessary to heat up and then you have to wait for cool down.

ulab, 2023-05-15:
If your printer is equipped with an MMU, there is no material loaded. It gets loaded when its needed and unloaded after each print. 

This menu wouldn't help in that case.

bkerler, 2024-01-26:
I wrote a PR #3709 to fix this confusing ui issue, now showing the right temperature in the preheat menu.
However I do like more a menu with the option to just heat up the heatbed or the nozzle or both as mentioned above, but that will take a bit more time for a proper implementation.

bkerler, 2024-01-27:
PR #3709 now adds a full menu with several options for preheating :)

bkerler, 2024-05-25:
Writing a comment here as it is still open and needs to be resolved.

CZDanol, 2024-07-30:
Internal ticket: BFW-5879