prusa3d/Prusa-Firmware-Buddy #1812: [FEATURE REQUEST] Confirmation of Filament change during printing

volkhen, 2022-01-10:
I got frustrated today when during printing I wanted to change speed and... The cursor jumped to one higher position to "Filament change" and immidiately started unloading the filament. This resulted in my frustration as it was transparent filament and after resuming printing there was clear mark of that moment. 

My proposal it to add a **confirmation stage** (YES/NO) after you push Filament Change like you have when you click Cancel Print. 

What do you think?

TerTerro, 2022-01-11:
If that option is "Optional"
Don't want to have to click additional Yes/No. Never had issue with it. As had issue with purge/change filament, when misclicked purge few times lol :) But need to be more careful.
So I don't like the idea of additional yes/no clicks

volkhen, 2022-01-11:
The problem is that while pressing the wheel sometimes it does unintended rotation moving the cursor. This happens regardless if you are careful or not. Imagine if that happens during long print.
well, imprementing it is not hard and the consequences of misclick are quite big - ruined print. OK, maybe not completely ruined but estheticly unpleasent ;)

closed 2024-06-18 by github-actions[bot]