prusa3d/Prusa-Firmware-Buddy #1821: [ENHANCEMENT] Filament change temps

DanidCH, 2022-01-13:
I’m wondering if for a filament change it wouldn’t be better to use the hottest of two temperatures (between last filament and new filament) during the purge. This would probably make the change easier.

Here is a specific case to illustrate how this would help: let’s say I used ASA for my print where hotend is set to 260c and I next want to use PLA, while purging the hot end will be set to 215c (recommendation for PLA) when there is probably some ASA left.

Probably setting the hotend temp to the higher temperature will help the purging process.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI]

zoltan-l, 2022-01-20:
Hi @DanidCH ,
thanks for your input, we will look at it

TerTerro, 2022-01-29:
Agree on this. I always do, load PLA after ASA, but specify I'm loading ASA, purging 3-5 times and then unload and load again same PLA with PLA temps. Would be great option to have, switching to purge at old filament temps.

JdogAwesome, 2022-02-26:
Couldn't agree more on this, would love to see this as a feature. Often when I'm printing PC and go to print PLA I will try to do it as quickly as possible to keep the nozzle temp high to get all the PC out of the hotend.

kozross, 2023-07-07:
It's been over 18 months since this was reported, and I just had a more current issue closed in favour of this one, along with two others in the last four months. Is there any ETA on this being dealt with? Given that we can't manually load filament into the MINI+ without going through this system, and I personally have clogged my printer a few times this way already, it'd be good to have a roadmap, or even an explicit statement that we shouldn't wait for a solution.

Prusa-Support, 2023-09-27:
I'm sorry, no estimated time available and I'm not sure this is in the current development plans.
The issue will stay open.

Michele Moramarco
Prusa Research

Tupson444, 2023-12-15:
One of the basic advices for 3D printing is e.g. "when changing between PLA and ABS, set the temperature for ABS". This is an obvious overlook that should be fixed.

mjablon1, 2024-01-12:
@Prusa-Support can you try again putting this into the development roadmap, please?

bkerler, 2024-01-13:
I wrote a PR #3681 for it. Only tested so far in the simulator, but it should work fine :)

Prusa-Support, 2024-02-16:
I'll try my best to escalate. 🙂
During the last year, our developers worked tirelessly and released more than a dozen firmware stable releases in this repository, and there is still a lot to do.

This issue will remain open as inspiration for future development but opening a Pull Request would probably be more effective. Thanks for it.

Michele Moramarco
Prusa Research

Zappes, 2024-02-17:
@Prusa-Support Please try to actually read the comments before answering in your usual way... @bkerler HAS created a pull request and actually linked to it.

Prusa-Support, 2024-04-09:
> @Prusa-Support Please try to actually read the comments before answering in your usual way... @bkerler HAS created a pull request and actually linked to it.

I even thanked for it.

.

As highlighted by our collaborator in the MK3 FW repository - https://github.com/prusa3d/Prusa-Firmware/issues/3114#issuecomment-1990881827 - switching between vastly different materials could actually be troublesome if neither is correctly removed first.
For example, PLA will probably carbonize and harden in no time if un/loaded at PC temperature.
A feature that automatically picks the higher temperature has to be carefully considered and thoroughly tested.

In most cases, a cold pull between material changes is very recommended.

Michele Moramarco
Prusa Research

KrisasMan, 2024-04-30:
I ended up in the wrong repository for my own version of this request.
(https://github.com/prusa3d/Prusa-Firmware/issues/4675)

My experience has been that if you keep PLA flowing the entire time it won't carbonize.  The key would be to have the printer start at the higher temp and then, after sufficient purging, drop the temp and continue purging to drop the temp to PLA temp.

JGeiger2024, 2024-10-17:
We really need something done to address this.  I have run into issues with MK4/MMU3 when trying to print PLA and TPU

Prusa-Support, 2024-11-28:
Material in a similar temperature range like PLA and TPU may not necessarily manifest temperature-related problems when loaded one after the other. However, flexible materials may be tricky sometimes so the problem is not necessarily relevant for this issue.

Flexible materials may potentially stretch, squeeze, and curlup on the way through PTFE tubes and inside the MMU module, so they require extra care.
Also, please note that officially supported materials in the case of _MK4/S with MMU3_ are PLA and PETG.
https://blog.prusa3d.com/development-diary-update-on-mmu3-for-the-mk4_85043/
Advanced users may be used in MMU only experimentally. Otherwise, please use standalone printer profiles.

I'm sorry for the lack of updates to this issue but the previously shared information still applies - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1821#issuecomment-2045584354. 
The issue will remain open for the time being but I recommend checking carefully the above-mentioned comment and its link.

Michele Moramarco
Prusa Research

Tupson444, 2024-11-29:
Changing between very different filaments regarding temperature (e.g. PLA and PC) indeed would be problematic if done directly (either PLA would carbonize and clog, or PC remains wouldn't be melted enough and also cause a clog). That is a good point and shouldn't be ignored.

But, very often filaments are changed between close ones in temperature, e.g. PLA and PETG, PETG and ABS, ABS and PC. In that case, it is better if purged at the higher temperature, and much quicker and easier than doing a cold pull in between.

Maybe add a setting in the configuration to be able to switch on "purge at higher temperature"? Or even better, to have a limit to prevent switching between filaments with too large temperature difference, say >30°C. In that case, if attempting to change e.g. from PC to PLA, the printer would show a warning, giving options "continue anyway" or "cancel", while suggesting to first do unload, cold pull, then load new filament (maybe even with a third choice that would perform unload, then guide the user to do a cold pull, then load).

What others think? Is this a good idea? I think it would solve the problem without causing a different problem, while the printer would also guide and warn users who are unfamiliar with the potential issue.

spooknik, 2024-12-17:
Recently upgraded my MK3S+ to a MK4S and I would just like to chime in and say I am having problems everytime I change from PETG to PLA even. Basically I get a 'soft clog' everytime it happens. So when swapping materials the awesome auto load function is broken for me. I need to manually set the hotend temp to 250 and manually extrude the PLA in. On the MK3S+ this was never a problem

I really hope some progress can be made on this soon, since the issue was started nearly 2 years ago. 

> What others think? Is this a good idea? I think it would solve the problem without causing a different problem, while the printer would also guide and warn users who are unfamiliar with the potential issue.

In my opinion the firmware should store the previous loaded filament and compare the temp vs. the current loading filament. If current filament temp < previous filament temp, apply a purge temp offset. So for example

PETG 240C to PLA 215C. Delta is 25C, purging temp should be 230 or so.

Tupson444, 2025-02-23:
not stale

albjerre, 2025-05-27:
Yes, we need this! I had the same issues with nozzle clogs on Prusa XL, before I made a workaround.

On my Prusa XL, when changing from a high temp filament to a low temp filament, I currently have to use a workaround where I select the high temp filament as the new filament. Then I change to the filament, purge three times and then initiate a new filament change where I select the low temp filament as the new filament. This is cumbersome and could easily be avoided if the firmware always used the temperature for the high temp filament when changing.