prusa3d/Prusa-Firmware-Buddy #1833: [BUG] Filament sensor turns off by itself on the printer

majedation, 2022-01-20:
**Printer type** - [MINI]

**Printer firmware version** - [4.3.4]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint** - [USB flash drive]

**Describe the bug**
The filament sensor option turns off by itself in the printer.
Each time I turn on the "Filament Sensor" option in the settings menu in the MINI, couple days later I check it again and see that it's off. I've asked in the facebook group and couple users had the same issue.

**Expected behavior**
The "Filament Sensor" option in the printer should not toggle on/off till the user toggles it manually.

zoltan-l, 2022-02-12:
HI thanks for info we are going to validate. 
Did it appear several times even after the printer reset?

majedation, 2022-02-12:
> HI thanks for info we are going to validate.
> Did it appear several times even after the printer reset?

Yes, even after turning off/on the printer.

murk-sy, 2022-03-11:
Can confirm this just happened to me too on my 2nd printer, and this is the first time. I only remembered to do a crash dump after I turned it on back, and it was identical to normal.

@majedation Likely unrelated, but is/was your Mini connected to the network? 

I'm probably chasing windmills, but I feel like some networking related overflow is the culprit in this and a few other weird bugs I've been having.

matt-chapman, 2023-05-06:
I am experiencing this issue on my new MINI+:

Firmware: 4.4.1+4194
Bootloader: 2.0.6

The issue does not occur when I downgrade the firmware to 4.4.0, which suggests the possibility of a regression.

My MINI is connected to a network via wired ethernet, no ESP module installed.

Interestingly, if I disconnect the printer from my network and reboot it a few times, the filament sensor issue does not occur.

I'm happy to provide dumps or more debugging info if needed - I would love to have this solved.

Prusa-Support, 2023-06-09:
Thank you for reaching out to us.

The issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3030 (apparently a recent duplicate) seems to clarify under what circumstances this issue occurs:
_the filament sensor gets automatically disabled **only if** USB drive and Ethernet cable are **both** plugged **while** rebooting_.

The last message above seems to match this thesis, but the ethernet connection doesn't seem to play a role for everyone.

Are other users in this thread able to reproduce and confirm this, trying all combinations with (and without) an Ethernet cable / USB drive?

Before testing, please notice that connecting an Ethernet cable may not be enough if you also have an ESP module connected: the ESP module may override and force the Wi-Fi connection (as speculated in issue 2961).

Michele Moramarco
Prusa Research

Prusa-Support, 2024-02-02:
The firmware was greatly improved and uniformed across all Buddy boards (MK4/XL/MINI) since 23 Nov 2023 with FW 5.1.0, rendering the issue potentially obsolete.

Please let us know if the problem persists.

Michele Moramarco
Prusa Research

matt-chapman, 2024-02-04:
Happy to say I have not had this issue since the update(s) :)

Prusa-Support, 2024-02-22:
Glad to hear that 🙂 and thanks for your feedback.
This issue will be closed now.

Michele Moramarco
Prusa Research

closed 2024-02-22 by Prusa-Support