prusa3d/Prusa-Firmware-Buddy #1836: [FEATURE REQUEST] Bowden length compensation setting

Lichtjaeger, 2022-01-27:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]
**Optional upgrades** - [Revo Hotend Upgrade]

**Describe the expected functionality**
That new experimental HW menu is nice. But could you please add a setting to adjust the extruder to nozzle distance to compensate for different hot-ends or Bowden lengths?

TerTerro, 2022-01-29:
Sounds like a great suggestion.

torkildr, 2022-02-02:
Haven't done this myself, but I would guess something like this would be pretty essential for a proper direct drive conversion.

GuidoHendriks, 2022-02-13:
This would be an amazing feature. Is it currently possible to achieve the same result using Gcode?

ardenpm, 2022-05-28:
Indeed this would be helpful. The E3D Revo upgrade has you cutting the bowden tube down a tiny bit (though it does also describe how to remake at the correct length), as a result when loading the filament reaches the nozzle a touch earlier and the extruder has a hard time of it.

Prusa-Support, 2022-06-06:
Hello! Sorry for the late response on this thread. Thank you for the feedback and comments, I brought this to our devs attention so that it can be further evaluated. In case of more info we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

SnickCH, 2022-06-30:
This would be graet. Then I could upgrade to an orbiter and just add the new lenght

closed 2024-06-18 by github-actions[bot]