prusa3d/Prusa-Firmware-Buddy #1838: [FEATURE REQUEST] nozzle changing setting

printerer, 2022-01-28:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - [MINI]

**Describe the expected functionality**
 an option for you to select that will retract the filament a set amount of distance to allow a nozzle change without the chance of hot filament causing any issues

TerTerro, 2022-01-29:
Sounds a nice idea. Like having menu, change Nozzle, that retracts filament and after change, you can set back to push filament in.

printerer, 2022-01-30:
> Sounds a nice idea. Like having menu, change Nozzle, that retracts filament and after change, you can set back to push filament in.

Yeah, exactly! Only thing is you can do it manually once the printer reaches a certain temperature… didn’t know that until after I posted

closed 2022-01-31 by printerer