prusa3d/Prusa-Firmware-Buddy #1861: [QUESTION] <Why is the master branch still version 4.3.3>

MinnesotaMean, 2022-02-07:
I am using prusaslicer 2.4.1 which takes advantage of some firmware improvements that version 4.3.4 of the buddy firmware also uses. 
I want to compile some changes from the master branch and realized that it is only on version 4.3.3. Just wondering why it is a version behind.

Thanks

zoltan-l, 2022-02-08:
Hi @MinnesotaMean ,
we know about it and we will synchronize master (which is a development branch) in short time period. Feel free to merge those differences into your fork. There are AFAIK 12 commits.

DRracer, 2022-03-30:
@MinnesotaMean the master branch has now all the features from 4.3.x. As I wrote elsewhere, please hold on with your compilation of the master branch (bbf file size and related bootloader changes)

closed 2022-03-30 by DRracer