prusa3d/Prusa-Firmware-Buddy #1872: [ENHANCEMENT] Add option when adjusting z-offset for build sheets to adjust all build sheets by a relative value

rcmaniac25, 2022-02-10:
Adaptation of https://github.com/prusa3d/Prusa-Firmware/issues/2905 for Buddy Board-based firmware.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - Mini/Mini+

**Is your suggested improvement related to an existing problem? Please describe.**
No, it's a general time saving feature (see description below)

**Describe the expected functionality**
If you have a single build sheet, the option to switch build sheets does not exist. If I change a nozzle or do some repair/maintenance on the extruder, there's a (good) chance I need to adjust the z-offset for the bed. This is generally easy.

Once you add additional build sheets, the option to switch sheets appears, and now when the same nozzle change/repair/maintenance occurs, each bed will probably need the z-offset adjusted. As a enhancement, allow adjusting all build sheets by a relative amount when adjusting z-offset. This should only show if there's more then one build sheet recorded.

Example:
- Smooth1: -0.543
- Smooth2: -1.122

I set to smooth1 and specify I want to adjust z-offset. I eventually get the following:
Do you want to use the current value?
Current: -0.543.
Default: 0.000
Click NO to use the default value (recommended)
[YES] [NO]

If YES is chosen, I would want the following:
Do you want to update all sheets?
Click YES to adjust all sheets by the same difference between old and new value.
Click NO to only adjust this sheet.
[YES] [NO]

If NO is chosen and the value resets to default, it should only effect the existing sheet.

If I choose YES for updating all sheets, then all sheets are adjusted by the relative change that occurred to Smooth1. So if I adjusted Smooth1 from -0.543 to -0.600, a difference of -0.057, then Smooth2 would also be adjusted from -1.122 to -1.179 (or I think it would actually save to -1.180 as precision is 0.002-0.003).

The purpose is to save time in adjusting build sheets when a system change occurs. It won't always be perfect, but it means you do a little bit of live-z adjusting instead of digging into a bed, loosing a first layer, or going "oh no, I forgot to adjust it" and needing to stop a print and adjust the z-offset on it's own.

Speaking of live z, it may be useful to have an additional option to live z adjust all sheets, but I don't think it's needed (and could add confusion if you're rushing to quickly adjust the first layer that you realized isn't as perfect as you hoped)

SnickCH, 2022-06-30:
I just had to replace the nozzle and I can now recalibrate all printsheets. With this feature I would only have to recalibrate one

Prusa-Support, 2022-07-16:
Thank you for your input.

Sounds clever to me : ) 
Not sure if the current firmware can support a calculation of the extra offset and a simultaneous update for all profiles. Maybe this will require a partial rewriting of the current feature or an additional separate feature.
I'd leave this up to the developers as inspiration for future firmware improvements.

Michele Moramarco
Prusa Research

closed 2024-06-18 by github-actions[bot]