prusa3d/Prusa-Firmware-Buddy #1875: [BUG] Octoprint in-print display not working

andywarburton, 2022-02-14:
**Printer type** - MINI

**Printer firmware version** - 4.3.4 (latest public release)

**Original or Custom firmware** - Original

**Optional upgrades** - Bondtech extruder

**USB drive or USB/Octoprint** Octoprint

**Describe the bug**
Since upgrading to 4.3.4 I no longer get the octoprint display on my prusa mini when connected to Octoprint. My Mini just stays on the printer home screen, even when there is a print in progress. This makes it impossible for me to adjust things like speed during the print. I tried creating a fresh install of Octoprint but still experience the same issue.

**How to reproduce**
Install 4.3.4, connect to Octoprint, try to print.

**Expected behavior**
I expect to see the "octopus" screen on the mini when connected to Octoprint.

andywarburton, 2022-02-15:
[image]

I hope this photo helps illustrate the issue a little more clearly. As you can see even though it's printing my mini is stuck showing the "home screen" and not the "printing screen" (neither the default one or the octoprint one)

JohnnyDeer, 2022-02-16:
Hi @andywarburton thank you for reporting. I tried it on few MINIs and all works correct. What version of Octorint are you using?

coaxial, 2022-02-22:
@JohnnyDeer Also running 4.3.4 and 1.7.3 here, it shows the Octopus screen for me.

hstag, 2022-06-30:
This also happened to me more times. Same versions. If I connect directly via octoprint web interface everything is ok. But when I use some app for it (octoapp - > octoprint) then this bug occurs sometimes.

Prusa-Support, 2022-07-16:
Thank you again for reporting.

Have you tried downgrading or upgrading the Octoprint version?
Have you tried downgrading the MINI firmware version?

Does it occur very erratically or you are able to recognize a pattern and reproduce the problem intentionally?
Maybe some logs from Octoprint of the last few operations, right before this occurrence, may reveal some clues.

Michele Moramarco
Prusa Research

AJErazzor, 2022-08-10:
same for me its hit and miss whether the serial print is detected on the screen however the print continues fine and is communicating with octoprint

closed 2024-06-18 by github-actions[bot]