prusa3d/Prusa-Firmware-Buddy #1888: [FEATURE REQUEST] <Hotend Fan Speed Control>

E456a, 2022-02-21:
The hotend fan on a Prusa Mini(+) is relatively loud. Since there is already a speed control for the print fan in the menu, I would suggest that an (automatic) control is also added for the hotend fan.

JustAnother1, 2022-05-15:
The hot end fan should not be loud. If it is loud that might be because one of the cables is too close? If that is not the case try unscrewing the screws that hold the fan ins place. But just a bit. Like half a turn or so. That way the vibrations from the fan are not pushed into the hot end assembly so the assembly will not resonate and therefore make the fan louder.
Also damaged or old fans can become loader before they fail.

Prusa-Support, 2022-05-16:
Yes, the hotend fan should not be too loud, and its speed should not be changed. Please contact us at info@prusa3d.com so we can check the noise.

Marina Curzi
Prusa Research

closed 2022-05-16 by Prusa-Support