prusa3d/Prusa-Firmware-Buddy #1890: [QUESTION] Feature toggle on the master branch

GatCode, 2022-02-22:
Hi everyone,
I would like to ask if there is some kind of feature toggle on the current master branch, so that I can disable some of the experimental features, get rid of the grey background on the menu items and somehow save some flash memory such that there is no issue with the ESP Wifi setup?
Basically my goal is to have a setup which is exactly like the official 4.3.4 release but with the ESP thing enabled. Is this possible?
Thanks!

MinnesotaMean, 2022-02-22:
The master branch is not on 4.3.4

GatCode, 2022-02-23:
This was not what I was asking

DRracer, 2022-02-28:
@GatCode please stay tuned, the new stable networking is really shaping up and once it reaches some level of maturity, we'll clean up the master branch and release the first tested and signed beta of FW 4.4. The plan is to do so in March 2022.

GatCode, 2022-03-01:
Aah ok thank you very much.

closed 2022-03-01 by GatCode