prusa3d/Prusa-Firmware-Buddy #1916: [BUG] Attempt to enter an invalid instruction set state

elronzo, 2022-03-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Mini

**Printer firmware version** - 4.3.4

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Sensor

**USB drive or USB/Octoprint**
  Octoprint

**Describe the bug**
  After changing PLA filament I got a bluescreen with the following error:
  TASK: displayTask. Attempt to enter an invalid instruction set state
  (bluescreen attached)

**How to reproduce**
  Not sure. Still occured after pressing the reset button but disappeared after switching the Mini off and power it on again.

[image]

JohnnyDeer, 2022-03-14:
Hi  @elronzo, thank you for reproting. 
Can you please provide more information and crash dump? It happened by changing filament while printing? Does blue screen appears right after change, or printer does some moves, or are you able to click in menu, before BSOD? Are you able to reproduce it?
Thank you

Thynix, 2022-03-28:
I've been able to reproduce this twice now.

It doesn't happen when changing filaments during a print - just via the menu.

It's happened when I turn on the power strip that both OctoPi and the printer are connected to, then start a filament change from the menu before OctoPrint connects. (If I try to do so after OctoPrint connects, selecting Tune > Change Filament doesn't open the filament change menu or perform the actions, though it does beep to acknowledge the wheel press.)

It's been too long since the first time for me to remember the precise sequence of events, but for today's happening:

1. Turn on printer / OctoPi via power strip
2. Select Filament > Change filament
3. Printer heats up and unloads the filament (OctoPrint must have connected as this was happening)
4. I select PLA (same as unloaded type) to load
5. Printer suddenly restarts, bootloader runs, firmware presents crash screen and does error beeps

[image]

Here's what OctoPrint had in its terminal:

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyACM0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6ca66d50, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv:  T:33.41 /215.00 B:21.80 /60.00 A:21.50 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:36.55 /215.00 B:22.54 /60.00 A:21.50 /0.00 @:127 B@:127 W:?
Recv:  T:39.57 /215.00 B:22.66 /60.00 A:21.50 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:42.11 /215.00 B:23.24 /60.00 A:21.50 /0.00 @:127 B@:127 W:?
Recv:  T:44.92 /215.00 B:23.87 /60.00 A:21.50 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:48.31 /215.00 B:24.26 /60.00 A:21.50 /0.00 @:127 B@:127 W:?
Recv:  T:51.36 /215.00 B:25.06 /60.00 A:21.69 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:54.18 /215.00 B:25.28 /60.00 A:21.94 /0.00 @:127 B@:127 W:?
Recv:  T:57.21 /215.00 B:25.85 /60.00 A:21.75 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:60.80 /215.00 B:26.36 /60.00 A:21.91 /0.00 @:127 B@:127 W:?
Recv:  T:63.67 /215.00 B:26.53 /60.00 A:22.00 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:66.64 /215.00 B:26.87 /60.00 A:22.00 /0.00 @:127 B@:127 W:?
Recv:  T:69.77 /215.00 B:27.53 /60.00 A:22.00 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:73.13 /215.00 B:28.04 /60.00 A:22.00 /0.00 @:127 B@:127 W:?
Recv:  T:76.16 /215.00 B:28.24 /60.00 A:22.41 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:79.15 /215.00 B:28.78 /60.00 A:22.37 /0.00 @:127 B@:127 W:?
Recv:  T:82.03 /215.00 B:29.09 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:85.32 /215.00 B:29.49 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv:  T:88.15 /215.00 B:30.18 /60.00 A:22.31 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:91.01 /215.00 B:30.44 /60.00 A:22.09 /0.00 @:127 B@:127 W:?
Recv:  T:93.77 /215.00 B:30.83 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:96.91 /215.00 B:31.30 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv:  T:99.61 /215.00 B:31.67 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:102.34 /215.00 B:32.21 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv:  T:105.05 /215.00 B:32.63 /60.00 A:22.87 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:108.13 /215.00 B:33.31 /60.00 A:22.91 /0.00 @:127 B@:127 W:?
Recv:  T:110.80 /215.00 B:33.54 /60.00 A:22.72 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:113.47 /215.00 B:33.96 /60.00 A:22.50 /0.00 @:127 B@:127 W:?
Recv:  T:116.11 /215.00 B:34.32 /60.00 A:22.81 /0.00 @:127 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:119.03 /215.00 B:34.90 /60.00 A:23.00 /0.00 @:127 B@:127 W:?
[... 84 lines cut ...]
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M21
Recv: ok
Send: M155 S2
Recv: ok
Send: M876 P1
Recv: ok
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3940
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor
```

Thynix, 2022-03-31:
It happened again today: this time it restarted and gave the error screen just after running the extruder to unload the filament.

Thynix, 2022-03-31:
If you need more information, I imagine I could make a video of it happening, perhaps with timestamps and timestamped OctoPrint terminal output. Would you like me to do that?

elronzo, 2022-03-31:
I will have a look today if my mini still holds that particular crash dump. If yes, I'll provide it here.

Thynix, 2022-03-31:
Here's my crash dump [dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/8387006/dump.zip)

meermanr, 2022-09-18:
I've just this happen on my Prusa MINI+:

0. Prusa MINI is operating standalone - OctoPrint has *not* connected to it to begin serial control mode
1. Filament -> Change Filament (select PLA, same as it was set before)
2. While nozzle and bed are heating, launch OctoPrint and connect to printer
3. Remove unloaded filament and prepare my next spool - then get distracted, so printer is in this state for some time (10min? 20min?)
4. Return my attention to the printer and answer "Is colour correct" with "No", some more filament in purged
5. Respond to next "Is colour correct" with "Yes"
6. Blue Screen appears

[image]

I was a little worried that I'd left it at prompt (4) for so long that it might turn off the fans.

I'm not 100% confident about when I connected OctoPrint - it _might_ have been between the first and second "Is colour correct" prompt.

Crash dump file: [dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9593792/dump.bin.zip)

meermanr, 2022-09-18:
I've been able to reproduce this:

0. Reset printer (.. as it happens, immediately after posting the crash dump above)
1. Filament -> Change filament
2. During the "Waiting for temperature" display: connect OctoPrint serial printing (log below)
3. Temperature is reached, filament unloads
4. Blue screen appears. AFACT the "Is colour correct" screen was _not_ displayed.

[image]

OctoPrint serial log:

```plain
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x999f1b38, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:90.90 /215.00 B:33.39 /60.00 A:37.06 /0.00 @:86 B@:127 W:?
Recv:  T:93.57 /215.00 B:33.70 /60.00 A:37.02 /0.00 @:87 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:96.16 /215.00 B:34.06 /60.00 A:37.06 /0.00 @:88 B@:127 W:?
Recv:  T:98.83 /215.00 B:34.53 /60.00 A:37.06 /0.00 @:89 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:101.97 /215.00 B:35.19 /60.00 A:37.06 /0.00 @:90 B@:127 W:?
Recv:  T:104.70 /215.00 B:35.52 /60.00 A:37.06 /0.00 @:90 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:107.40 /215.00 B:35.94 /60.00 A:37.06 /0.00 @:90 B@:127 W:?
Recv:  T:110.05 /215.00 B:36.29 /60.00 A:37.06 /0.00 @:91 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:113.19 /215.00 B:36.67 /60.00 A:37.06 /0.00 @:92 B@:127 W:?
Recv:  T:115.74 /215.00 B:37.02 /60.00 A:37.10 /0.00 @:94 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:118.34 /215.00 B:37.38 /60.00 A:37.21 /0.00 @:95 B@:127 W:?
Recv:  T:120.99 /215.00 B:37.85 /60.00 A:37.32 /0.00 @:96 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:124.20 /215.00 B:38.29 /60.00 A:37.35 /0.00 @:96 B@:127 W:?
Recv:  T:126.83 /215.00 B:38.67 /60.00 A:37.35 /0.00 @:97 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:129.36 /215.00 B:39.02 /60.00 A:37.35 /0.00 @:99 B@:127 W:?
Recv:  T:132.06 /215.00 B:39.33 /60.00 A:37.35 /0.00 @:99 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:135.29 /215.00 B:39.92 /60.00 A:37.35 /0.00 @:99 B@:127 W:?
Recv:  T:137.86 /215.00 B:40.17 /60.00 A:37.43 /0.00 @:100 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:140.33 /215.00 B:40.53 /60.00 A:37.35 /0.00 @:102 B@:127 W:?
Recv:  T:143.16 /215.00 B:40.83 /60.00 A:37.35 /0.00 @:101 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:146.37 /215.00 B:41.36 /60.00 A:37.54 /0.00 @:101 B@:127 W:?
Recv:  T:148.93 /215.00 B:41.71 /60.00 A:37.65 /0.00 @:103 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:151.57 /215.00 B:42.06 /60.00 A:37.65 /0.00 @:103 B@:127 W:?
Recv:  T:154.30 /215.00 B:42.41 /60.00 A:37.65 /0.00 @:103 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:157.26 /215.00 B:42.83 /60.00 A:37.65 /0.00 @:105 B@:127 W:?
Recv:  T:159.65 /215.00 B:43.24 /60.00 A:37.65 /0.00 @:107 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:162.41 /215.00 B:43.57 /60.00 A:37.65 /0.00 @:107 B@:127 W:?
Recv:  T:165.04 /215.00 B:43.86 /60.00 A:37.72 /0.00 @:107 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:167.96 /215.00 B:44.32 /60.00 A:37.65 /0.00 @:109 B@:127 W:?
Recv:  T:170.39 /215.00 B:44.71 /60.00 A:37.94 /0.00 @:110 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:173.17 /215.00 B:45.05 /60.00 A:37.65 /0.00 @:109 B@:127 W:?
Recv:  T:175.79 /215.00 B:45.35 /60.00 A:37.94 /0.00 @:110 B@:127 W:?
Recv: echo:busy: processing
[... 62 lines cut ...]
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M155 S2
Recv: ok
Send: M876 P1
Recv: ok
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3997
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor
```

Crash dump: [dump B.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9593823/dump.B.bin.zip)

closed 2024-06-18 by github-actions[bot]