prusa3d/Prusa-Firmware-Buddy #1919: [FEATURE REQUEST] Add Filament to Prusa Mini Display

Toastinator, 2022-03-14:
Im Using PA-CF (290c) Filament and The Highest Filament Type is PC at 275. So I am unable to Load it Normally. I have to Do everything Manually. Adding a Filament to the Filament Menu Would Streamline this Process, And allow the Use of LOAD/UNLOAD.

People Who Use Filaments Like PLA+ that uses a High Temp then PLA would benefit. 

My issue is not with the Operation of the Printer, I know that the Slicer Determines the Print Temp. My Issue is with Loading/Unloading Filament on the Prusa Mini. I assume that is a Firmware Change. Would Really be Nice to Save Weird Custom Filaments to the Machine.

murk-sy, 2022-03-14:
+1. The same issue occurs when using first layer calibration - you have to use either PC or calibrate using a different filament type.

Custom filaments may be the best option since Prusa might not want to list filaments that aren't officially supported. But at that point we're talking about stuff that would be best managed from the slicer, a central print farm controller, or via the WUI.

KlausKraemer, 2024-02-07:
Please not only for the MINI!

Also the XL (and probaly all printers running this firmware) lack a possibility to add a new filament to the filament list. Maybe that additional list could be located on the USB Drive - a simple CSV would do.

closed 2024-06-18 by github-actions[bot]