prusa3d/Prusa-Firmware-Buddy #1930: [FEATURE REQUEST] Default option after a print should be HOME

Area5142, 2022-03-18:
Prusa Mini+
Firmware version: 4.3.4 and all earlier

**Describtion:**
After a print the dialog with the options _Tune/Reprint/Home_ is shown with _Reprint_ as the default:
[image]

In 99% of cases I don't want to reprint, but want to print another model and _Home_ should be the default.

Accidentally pressing the selector starts a reprint and it is cumbersome to back out of the operation and select a new print.

**Describe the expected functionality**
I expect the default operation after a print to be more safe like the _Home_ choice.
The current _Reprint_ choice is more suited if the printer is to be used primary in a printing factory, where a lot of equal parts is mass produced.

After all, the just printed part has to be removed and bed prepared for the next print before _Reprint_ makes sense to use - too easy to start a print before the previous print is removed...

murk-sy, 2022-03-21:
The assumption is that if you'll be starting a different print, you will (at least currently) most likely remove and replug the USB storage to get a newly sliced print, which would take you to the home screen anyway.
Not sure how common that type of use is, but I am *much* more likely to reprint the current gcode file or remove the USB storage than I am to print an older gcode file.

Area5142, 2022-03-21:
I never remove my USB storage as I am using a Toshiba-FlashAir and manages the print files over WiFi.
No need to remove the USB storage when I can send the next print file over WiFi while the current is printing.
Still a default of _Home_ after a print is finished would be more safe option and avoiding starting a reprint by accident.

vanboom, 2022-10-31:
From the screen shown, clicking "Home" does not take you back to the home screen, instead it browses over to the storage listing for the next print.  At least half of the time we are changing filaments, or submitting print jobs remotely from PrusaLink.  Clicking the "Home" selection from this screen should actually take you to the home screen.

Prusa-Support, 2022-11-03:
Hello! Thank you for the post and for the comments, sorry for the reply being so late from my end. In the meantime, I put this to our devs attention, both the aspect on the original post and the latest comment, it will be further reviewed from our devs. 

Alessandro Pantaleo
Prusa Research

closed 2024-06-18 by github-actions[bot]