prusa3d/Prusa-Firmware-Buddy #1967: [BUG] WatchDog Crash when running XYZ test and with USB plugged in

Madmatt06, 2022-03-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches. *Note There was another bug report that had a similar issue but it was on a older version, didn't mention anything about a USB drive seeming to cause it, and had to do with the fan testing after the XYZ testing succeeded.

**Printer type** - Mini+

**Printer firmware version** - 4.3.4

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
  When running the XYZ test with a USB drive the screen freezes on the test screen with nothing happening and then the printer resets with a WatchDog error. I checked all the connections and everything is plugged in and telling it to home does not crash it. I was able to reproduce it multiple times in a row with the same results. However upon disconnecting the USB drive (no restart necessary) the printer no longer crashes. The only file on the USB drive is not corrupted and does print correctly.

**How to reproduce**
  Plugin USB drive (preferably with the same file using the disk image/file included) and run the XYZ test

**Expected behavior**
  Printer freezes then restarts with a WatchDog Error

**G-code**
  G-Code on USB during time of crash: 
[Body1_0.25mm_PLA_MINI_3h19m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/8377806/Body1_0.25mm_PLA_MINI_3h19m.gcode.zip)
Disk Image of USB drive During Crash:
[Crashing USB drive.dmg.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/8377815/Crashing.USB.drive.dmg.zip)

**Crash dump file**
  Crash Dump created from crash:
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/8377820/dump.bin.zip)

**Video**
  Crash with USB drive connected:
  [Video](https://youtu.be/3fmXLjPWw3o)

GithubUser99999999, 2022-12-16:
I experienced the same bug for the first time now after I changed the nozzle to 0.6 mm and therefore had to run the self test again since months. It constantly gave me a crash plus watchdog error until I removed the USB drive and then it worked. Firmware is 4.4.0

Prusa-Support, 2023-01-17:
Thanks for reporting, I'm unable to reproduce the issue though.
The gcode doesn't seem to play a role here. Honestly, I don't see how that can affect the XYZ Test but I tested this anyway, and I couldn't reproduce the issue.

A constant concern so far is the USB flash drive itslef. It may be corrupted or faulty.
Please try other USB flash drives formatted FAT32. If the issue doesn't follow a specific pattern, even with other USB flash drives, it would be difficult to reproduce.
Please test other USB flash drives, and consider closing the issue if the problem doesn't persist.

Michele Moramarco
Prusa Research

murk-sy, 2023-02-14:
I started making a new issue but only later found out there was already one - might as well post here.
I believe this is just one of the many issues that are caused by USB drive corruption or some other issues related to it.
@Prusa-Support 

### [BUG] XYZ Self test causes watchdog reset if USB storage is connected (and possibly corrupt)

**Printer type** - MINI+
**Printer firmware version** - 4.1.1

**Original or Custom firmware** - Original
**Optional upgrades** - Filament Runout Sensor
**USB drive or USB/Octoprint** - USB Flash drive

**Describe the bug**
When trying to run XYZ Self test in the calibration menu, the calibration menu opens (axes are listed), but nothing happens. After a few seconds, the printer restarts, and after loading the bootloader, the printer throws a WATCHDOG RESET error.

Note that Windows volume repair tool does not sufficiently fix the storage to be able to run the test. Additionally, while there are FOUND.x folders on the drive, there are no files with corrupt names for example.

**How to reproduce**
1) Insert a (problematic?) flash drive.
2) Run XYZ self test in the calibration menu.
3) The printer locks up, restarts, and throws WATCHDOG RESET. **Note:** The file `test_<serial>_xyz.txt` is created, but is empty.
4) Restart the printer

5) Unplug USB storage
6) Start XYZ test again
7) The test runs normally

8) Plug in a freshly formatted USB storage device
9) Start XYZ test again
10) Although it takes a little bit to start, the test runs normally

**Expected behavior**
a) Run test normally, and/or
b) Notify user about USB storage issues

**Crash dump file**
[2023-02-14 xyz self test watchdog reset - dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10735746/2023-02-14.xyz.self.test.watchdog.reset.-.dump.zip)

**Extra info**
If requested, I will create a binary image of the flash drive and mail it in, as I've already done once before. Then the developers should be able to flash that exact data to a standard included Mini USB storage and reproduce it.

Prusa-Support, 2023-02-23:
Thanks for the additional information.
We have more confirmations about a possibly corrupted USB flash drive.

I believe that the Windows Repair Tool is very a powerful tool but, from my experience, it may not be a definitive solution.
In other words, corrupted flash drives may be a cause of very unpredictable and erratic behaviors.

@murk-sy. Storage corruption is difficult to reproduce intentionally. We will let you know if our developers find that a raw image can be of help.

Michele Moramarco
Prusa Research

closed 2024-06-17 by github-actions[bot]