prusa3d/Prusa-Firmware-Buddy #2018: [FEATURE REQUEST] Ability to self-sign FW to test without removing appendix

Danil0v3s, 2022-04-12:
**Printer type** - [MINI]
**Optional upgrades** - [e.g. Filament Runout Sensor] - N/D

**Is your feature request related to an existing problem? Please describe.**
Not being able to compile our own firmware without voiding the warranty seems a bit counter intuitive with the open-source approach.

**Describe the expected functionality**
One should be able to sign their firmware with one of the many many methods available so they could test the firmware on their own machine.

murk-sy, 2022-04-14:
Allowing self signed certificates would make the appendix completely useless in the first place.

The appendix exists for 2 reasons (based on my own conclusions, not some official statement):
1. to limit Prusa's liability in case of a fire or other external damage caused by badly configured printer
2. to discourage users from flashing firmware they "found on the internet and it's supposedly better" by requiring the fairly arduous and scary procedure of opening the electronics box and breaking the board

Think of it this way: the warranty on the printer for consumers in the EU is 2 years and outside 1 year. It makes no difference if you break the appendix yourself after this period runs out.

While I agree having the ability to flash custom firmware without that process would be nice to have, I completely understand that they need to draw a line somewhere. Perhaps they considered historical data on the i3 claims caused by custom firmware. It could be that the Mini will stay the exception. Only time will tell.

Danil0v3s, 2022-04-14:
Self-signing a build is only part of the process that could be done to allow users like me to start contributing.

Approval based applications to have a certificate or anything like that, having to apply to get a certificate would filter such cases of flashing firmware found on the internet.

murk-sy, 2022-04-14:
Certificates like that could not be revoked, which is again just a massive liability for vetting developers and their code.
It's doable by having some sort of a remote build service hosted by Prusa that spits out a firmware file - which also gives warnings before flashing - but that just seems like a lot of extra work for a subset of a subset of users.

I think equating non official firmware = warranty breaking is reasonable, because no custom firmware developer would guarantee part replacement or anything along those lines for their own major screw up.

To be clear, that's a theoretical edge case, but I guarantee the first custom "official-ish" firmware to be out would raise the max temperatures of the bed and the extruder. Probably not something that Prusa would sign off on.

JustAnother1, 2022-04-16:
Another option would be to create the option to create firmware that is locked to the own printer. No vetting of developers would be needed. Anyway allowing users to do more will be some effort for prusa. But I thought that prusa, as an open source company, would understand that it is worth the extra effort.

If someone wants to help prusa by contributing code, or just by testing the code of other community members, then that is only possible by breaking the PCB. That is why we see so little community involvement with the mini.

If I intentionally destroy the printer then that is not covered by the warranty. So prusa will already check on warranty claims if I took a hammer and destroyed the printer or if it failed due to some supplier or assembly issue. Stating that they take the appendix status into account in this check should be enough. With a statement like that they could even ask the experienced users to show support by breaking the appendix and testing new stuff. Making clear that the appendix helps new users to not do possible dangerous stuff, but on the other side stating that removing it allows experienced users to unlock the full potential would probably change the situation. This "If you break the appendix then you are on your own" is too scary for too many people.

But maybe prusa just starts moving away from open source, like others did: Makerbot, Ultimaker,... :-(

Ondrashek, 2022-05-20:
I would like to have these (link below) firmwares signed to trying latest commits. But printer can't recognize firmware file. :-)  

https://holly.prusa3d.com/job/Prusa-Firmware-Buddy/job/Multibranch/job/master/

Prusa-Support, 2022-05-25:
Hello! Sorry for intervening on this thread so late. Unfortunately, in this regard, I have to confirm the info about custom firmware from this guide: https://help.prusa3d.com/article/flashing-custom-firmware-mini_14, it wouldn't be possible to self-sign firmware without breaking the appendix. This doesn't go against the Open Source policy: everyone would have access to the FW, can modify it, flash it to a printer, but it has to be considered "at own risk and peril". 

To better address the warranty aspect: we'd still honor the warranty of replacement parts as long as there is no suspicion of the custom firmware causing the problem. For the sake of example, in case of an intermittent Mintemp Bed error caused by a broken heatbed thermsitor wire, even in case of custom firmware installed, after confirming the cause being the thermistor wire, we'd still consider the thermistor within warranty. 

Closing the issue based on this info, but of course in case of anything more to review and clarify, feel free to reopen it at any time, and we can further review these aspects. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-25 by Prusa-Support