prusa3d/Prusa-Firmware-Buddy #2116: [BUG] <Enter comprehensive title>

E456a, 2022-05-14:
Printer: Prusa Mini+
Firmware: Original 4.3.4

Bug:
  When the z-axis is moved to almost! (not quite) all the way up and you then go to change filament, the motor tries to rotate beyond the functional range

How to reproduce:
  Move the z-axis to almost the top by using the menu: Settings -> Move Axis -> Z
  Then go to Filament -> Change Filament
  wait for temperature

  Afterwards the z-axis motor will make a strange noise, because it tries to travel further upwards

Expected behavior:
  Two possibilities:
    1. The motor turns only up when the nozzle is above say 15cm
    2. The motor turns only up if there is enough space to do so

JustAnother1, 2022-05-15:
I have seen the same effect.

I assume that the Mini does not keep track on where the nozzle is and just always moves up on filament change. The Z speed is then to slow moving for the Trinamic driver to detect the end of the axis.

That would explain the behavior, but would also mean that this is hard to fix. But I might be wrong, didn't check the code,...

E456a, 2022-05-15:
It should not be difficult to make a simple check of the nozzle height before moving up or am I missing something?

JustAnother1, 2022-05-15:
There is no "position sensor" on the Z Axis. The only way to know the position of the Z axis is to home it. That is why every print starts with homing the axis.

If you printed a tall object (Z Axis is near to top position) and then shut down the printer. The printer then forgets the position of the Z Axis. Also manual movement of the axis is possible. When switching on the printer it does not know the position of the Z Axis. And it can not home, as the big print might still be on the bed. If you now do a filament change the up movement will happen and the printer has no chance to know that it now runs into the upper end of the printer.

I don't know how to fix that, other than adding a max Z end switch to the printer.

Then there is another situation, the probably more regularly happening one: You start a tall print and close to the end of the print a filament change is in order. As the Z axis has been homed at the start of the print, the printer could now count the number of steps it took on Z to then know that it is near to the top. In this situation slamming into the top end could be avoided by the firmware, as the printer has the chance to know where it is.

Prusa-Support, 2022-05-17:
Hello! Thank you for your message. I changed the tag for now given the info at hand, this will be brought to our devs attention. Were there more issues that you might have noticed stemming from this "rotation out of range"? 

Alessandro Pantaleo
Prusa Research

closed 2022-05-18 by Prusa-Support