prusa3d/Prusa-Firmware-Buddy #2127: [FEATURE REQUEST] add a prometheus metrics endpoint

weikinhuang, 2022-05-21:
**Printer type** - [MINI]

It would be great to have some prometheus metrics from the printer directly, currently I'm using octoprint prometheus exporter plugin https://plugins.octoprint.org/plugins/prometheus_exporter/ to generate the metrics page. But since I'm using a rp0 w2 and running into some wifi issues with disconnects, it would be nice to have it directly on the printer itself.

deltamualpha, 2022-06-20:
@weikinhuang I wrote an exporter that runs on a network-connected machine and scrapes the Prusa-Connect json endpoint. It's pretty solid! https://git.yetaga.in/alazyreader/prusa-connect-exporter

RichiH, 2022-08-07:
Hi @Prusa-Support ; as you're using Grafana internally, having a Prometheus-compatible `/metrics` endpoint would most likely help with your own data collection as well.

Prusa-Support, 2022-08-13:
Thanks for the suggestion.
We will keep this issue open and leave this up to the developers.

Michele Moramarco
Prusa Research

weikinhuang, 2024-05-22:
This is still useful to have

danopernis, 2024-08-19:
Bad bot!

pstrobl96, 2024-09-13:
I've talked with @gouthamve and the idea is that it cloud be possible to create second handler to expose /metrics endpoint. I have to do more research about this how much work it will take in the firmware code however this is the idea. If exposing /metrics endpoint is no go, then implementation of [Remote Write](https://prometheus.io/docs/specs/remote_write_spec_2_0/) is also possibility. IWith Prometheus metrics we are not aiming to extreme precision we want to have observability for monitoring, it's not meant for development as "_syslog_" implementation.

It will really help me because I could effectively stop development of [prusa_exporter](https://github.com/pstrobl96/prusa_exporter).

RichiH, 2024-09-13:
I would strongly advise to support a `/metrics` endpoint. It will be the least work, by far, for the maintainers of the firmware.

All the heavy lifting, complexity, service discovery, labeling and overrides, collection, pipelining, etc pp Just Works(TM) as it's being taken care of by existing software which focuses on this.

danopernis, 2024-09-13:
The `/metrics` endpoint is a preferable option in Prometheus ecosystem, but it remains to be seen if this will be implementable given memory constrains. We would probably need to ditch existing metrics subsystem as we can't really afford having two similar systems in place.

I personally like Prometheus's approach of "don't call us, we call you", I am just cautious when it comes to implementing it on our SoC. Especially MINI is really on the edge with available FLASH space. If this was to be implemented, it would have to be programmed in very streamlined fashioned and I don't generally recommend it as a first issue if you are new to embedded programming and this particular firmware.

RichiH, 2024-09-13:
Is dropping the old system and option? And, while it sucks, is dropping the Mini an option?

pstrobl96, 2024-09-16:
I'm not sure that is possible to drop old metrics system. Development metrics works with nanosecond precision which is not possible for Prometheus. I could imagine having two builds, one with "syslog" metrics and second with Prometheus but that would be too much hassle imho. I will rather drop Mini support instead.

RichiH, 2024-09-16:
NB: Any Prusa-internal people reading this, ask @pstrobl96 for the recording of the Prusa-interal talk I did on this talk in June(?) of 2024 in your Prague HQ.

Syslog is not geared towards metrics, it's built for logs/events.

The way this problem is commonly solved across **_all_** established industries is by splitting areas of concern:
* The normal operation is metrics-first to metrics-only. Most data fall into this category and they do not need high time precision. Common examples are power usage, network traffic, or vehicle speed
* Errors and other events which have a specific time of occurence are emitted as logs with their own timestamp
* Special situations are treated differently: A ringbuffer (or similar) keeps track of parameters at high time fidelity continuously. In the case of a relevant event the buffer is externalized. Common examples are the detection of transient spikes in power networks by Janitza etc., Microburst detection by Arista etc., and crash detection by BMW etc.

This clean separation of areas of concern minimizes overall resource usage while maximizing the information density of the data which is
1. generated,
2. emitted,
3. transmitted,
4. subsequently stored, and
5. retained longer term

with possible filtering steps at every stage. If that system looks the same between e.g. CERN's LHC and Grafana LGTM stack, that's no accident. My reply above focuses on 1. and 2.

danopernis, 2024-09-16:
I agree that mixing logs with metrics is wrong. I am just saying that we will be dropping some compatibility with existing metrics and I am warning you that implementation on embedded systems might not be as easy as you imagine it coming from other systems. I am not saying it is impossible and I am not saying I am against it :slightly_smiling_face:

RichiH, 2024-09-16:
Yeah, the complexity of the code I ran was certainly lower and outside a few Arduino playthings, the lowest power I ever touched was ESP32. I know it's hard enough for me to stay away from it.

I certainly have _opinions_ on having a WiFi board which is more powerful than the mainboard in my printer, yes ;)

It's a pity that the ESP32 is not in the data path for wired network, else offloading the HTTPs etc to it might be an option.

weikinhuang, 2025-11-05:
i think this is still important