prusa3d/Prusa-Firmware-Buddy #2129: [BUG] Firmware crashes when Load Filament is in "Color Correct?" state and octoprint connection is initialized.

drg407, 2022-05-22:
**Printer type** - MINI+

**Printer firmware version** - 4.3.4

**Original or Custom firmware** - Original

**Optional upgrades** - FRS

**USB drive or USB/Octoprint** - Octoprint

**Describe the bug**
  Firmware crashes when Load Filament is in "Color Correct?" state and octoprint connection is initialized.

**How to reproduce**
  Load filament and leave on "Color Correct" state
  From Octoprint, initialize connection to printer

**Expected behavior**
  Firmware should reject/wait for serial connection to complete and not crash

**G-code**
 N/A

**Crash dump file**
  
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/8749566/dump.bin.zip)

**IMAGE**
  
[image]

Prusa-Support, 2022-05-24:
Hello! Thank you for the report, sorry for the issue. So would it occur only if we initialized Octoprint during that specific operation, but not if we initialized it before that? I'll report to our devs with this detail, we'll look into it. 

Alessandro Pantaleo
Prusa Research

drg407, 2022-05-24:
Correct, I've only been able to reproduce the issue when I leave the printer on "color correct?" after loading filament, then from my machine click the connect button in octoprint. Let me know if you need anything else.

Prusa-Support, 2022-05-25:
Thank you for the confirmation! After reviewing this aspect with our devs, I sadly have to inform we'd have to consider this as low-priority for the moment, mainly because we'd need to focus on more critical bugs at this time, and in part also because of not having official Octoprint support for our MINI. 

For the moment, I can only suggest the workaround of initializing Octoprint at another moment, before or after the loading procedure. Thank you again for the info and for the report! 

Alessandro Pantaleo
Prusa Research

drg407, 2022-05-25:
No worries, throw it in the backlog. 🙂

closed 2024-06-17 by github-actions[bot]