prusa3d/Prusa-Firmware-Buddy #2130: [BUG] Failed Prints and failed fan tests with a working fan

tracyjr, 2022-05-23:
**Printer type**
MINI+ (factory assembled)

**Printer firmware version**
prusa3d_fw_4_3_4_MINI.bbf

**Original or Custom firmware**
Original

**Optional upgrades**
Filament Runout Sensor

**USB drive or USB/Octoprint**
USB

**Describe the bug**
Watchdog Alert during prints. If reprinting the same model, will give Alerts at different stages of the print.
Watchdog Alert always occurs when using CALIBRATION>Test FANs
Fan Test is always successful when using CALIBRATION>SelfTest

**How to reproduce**
Run the tests mentioned above

**Expected behavior**
I expect successful prints when there are no issues with hardware, models or support structures, filament intake, etc.
I expect  CALIBRATION>Test FANs to show successful when fans are working correctly.

I've printed about 10 times since updating to 4.3.4 .... half of the prints failed. I have never had a single print fail before updating.

Prusa-Support, 2022-05-24:
Hello! Thank you for the report and sorry for the issue. If we were to temporarily downgrade to 4.3.3 now, would the issue still persist? Old versions: https://help.prusa3d.com/downloads/mini-2/firmware, procedure: "Alternative method - older firmware release" from this guide: https://help.prusa3d.com/article/firmware-updating-mini-mini_124784. 

Another aspect, though low suspects given the issue occurred only after firmware upgrade, what is the room temperature on the printer room? Any air currents such as an A/C unit or an open window, etc.? 

Let me know, we'll review it from here and make this right! 

Alessandro Pantaleo
Prusa Research

tracyjr, 2022-06-04:
Forgive the delayed response. I believe the problem was me. I turned on the MINI+ before I put the USB into it. Since then, I’ve turned off the printer any time I had to remove the USB to add files and only turned it on after putting the USB back in. This is not something that interfered with printing before, but I can live with it.

My printing area is usually around 78ºF (25.6ºC).  We cannot open windows where I live, entirely too hot outside and the A/C vent does not blow toward this side of the room.

Thanks for responding so quickly. I’m sorry I didn’t realize this before writing.

Sincerely,

Tracy Hendry, Jr.

closed 2024-06-17 by github-actions[bot]