prusa3d/Prusa-Firmware-Buddy #2148: Firmware Beta Wifi Modul

PhilippeGalid, 2022-05-25:
Thanks for your hard work,when will come the Beta Firmware with the new Bootloader for the Wifi Module ?

Thanks Philippe from Vienna

closed 2022-05-25 by PhilippeGalid

Jerzeek, 2022-05-27:
Why was this closed?

Erik262, 2022-06-29:
It was promised almost 3 years ago, that we will get a WiFi function for the MINI, and still no official release. But instead a new XL with WiFi. Other companies like Anker, Bambu lap and co. invented a whole new printer with a LIDAR sensor in that time. We wait here for a ridiculous WiFi connection. I'm really disappointed.

Prusa-Support, 2022-07-02:
Hello! Thank you for the post and for the comments, sorry for the wait on releasing the WiFi feature - it is still currently being worked on, sadly no timeframe for release yet but it will happen! 

Alessandro Pantaleo
Prusa Research