prusa3d/Prusa-Firmware-Buddy #2155: [QUESTION] boot vs noboot bbf firmware image

osresearch, 2022-05-27:
I'm having trouble finding any documentation on what the `boot` and `noboot` firmware versions are and which of the many files is appropriate for flashing onto a [Mini with the appendix removed](https://help.prusa3d.com/article/flashing-custom-firmware-mini_14).  Several `.bbf` files are built by the script and the `README` doesn't mention which to use.

```
% python3 utils/build.py --generate-bbf
....
Building finished: 2 success, 0 failure(s).
 mini_release_boot   SUCCESS
 mini_release_noboot SUCCESS
% ls build/*/*.bbf
build/mini_release_boot/firmware.bbf
build/mini_release_boot/firmware_update_pre_4.4.bbf
build/mini_release_noboot/firmware.bbf
build/mini_release_noboot/firmware_update_pre_4.4.bbf
build/products/mini_release_boot.bbf
build/products/mini_release_boot_update_pre_4.4.bbf
build/products/mini_release_noboot.bbf
build/products/mini_release_noboot_update_pre_4.4.bbf
```

osresearch, 2022-05-28:
I decided to yolo it and tried some of the different images.  What seemed to work was first to load a clean build of `build/products/mini_release_boot.bbf`, which updated the bootloader to 2.0.2.  Then when I loaded my modified version of the firmware with `CLIENT=YES`, it made it past the verifying screen and I can send network jobs to my mini.

So what I *think* I figured out from digging into the source is that the `boot` version of the firmware has the updated bootloader binary bundled with it and the code to do the update. Meahwhile the `noboot` version does not, which gives it extra space for additional features, but if the device has an older bootloader then the firmware might not work if it tries to use a newer feature.

If this is right, it would be wonderful to have a section in the `REDME` that explains the various generated files and which ones are appropriate for which uses.

zoltan-l, 2022-06-08:
Hi,
yes you are right.
the noboot frimware is used by developers to load the FW into machine, where there is no boot loader in the processor installed. The FW is loaded via other means than USB. The advantage of this is, that the debug programs can have more flash memory demand than standard release FW. The noboot cannot be loaded via USB stick.
The built products to be used are in the build/products/ subdirectory

For your purposes please use the boot .bbf files the bootloader can deal with.
Regarding the 4.4 version it will be able as you already discovered to update the bootloader as well. But this is still not official version, so any changes are on your discreete responsibility.

Prusa-Support, 2022-07-09:
Thank you for reaching us.

Can the issue be closed now?
Please let us know if you have any further doubt or questions.

Michele Moramarco
Prusa Research

closed 2023-07-26 by Prusa-Support

3DPGVA, 2023-09-27:
> I decided to yolo it and tried some of the different images. What seemed to work was first to load a clean build of `build/products/mini_release_boot.bbf`, which updated the bootloader to 2.0.2. [...]
can you please share the mini_release_boot.bbf file

kohend, 2026-03-13:
Can this be please be added to the README? I flashed like 3 times until I thought to try that.