prusa3d/Prusa-Firmware-Buddy #2161: [QUESTION] <Enter comprehensive title>

Lino77, 2022-06-01:
Is it possible to rotate the display 90 degrees? if so where exactly in the code

Prusa-Support, 2022-06-03:
Thanks for reaching us. 

No, I'm sorry, the LCD's interface can't rotate. I'm afraid this would require a substantial GUI re-work.

This sounds like a nice suggestion though. I'll label this as a 'feature request' for the 'user interface' so that the developers can take notes and find inspiration for further improvements.

If you are satisfied with the answer, you can close this issue.

Michele Moramarco
Prusa Research

Lino77, 2023-03-18:
Is it now possible to change the display from vertical to horizontal for the Mini, after which the XL and the Mini partly share the source, and this has its display horizontal, it shouldn't be a problem

Prusa-Support, 2023-03-29:
I'm sorry. Even if the Prusa XL is based on MINI's firmware and menus, it is a very different product requiring a substantially different firmware. It can't be easily adapted for MINI.

The release of Prusa XL is not really relevant for this issue: it doesn't change the fact this feature request would require a specific and thorough rework of the User Interface.

Michele Moramarco
Prusa Research

arekm, 2023-03-29:
Is XL firmware or github ? Not seeing it (or looking in wrong places). (I was assuming it's there due to GPL and XLs shipped to some people already)

closed 2024-06-17 by github-actions[bot]