prusa3d/Prusa-Firmware-Buddy #2169: [FEATURE REQUEST] Make Sheet Z-Offset Values Visible Somewhere

RAMilewski, 2022-06-07:
In the release notes for Firmware 4.3.4 the last paragraph reads:

_"In order to rename the profiles, a factory reset must be performed. Before this operation is performed, it is recommended to write down the current “Live adjust Z” value and use it later during recalibration"._

...but unless I'm missing something, those values for the various sheets are not visible anywhere in the U/I without re-running first layer calibration for each sheet.

Adding a page to the **Info** screens with this information would be most helpful.

Prusa-Support, 2022-06-07:
Thank you for reaching us.

The developers tend to include as much information as possible in the release notes, but they should possibly avoid lengthy descriptions if not strictly necessary.

No worries though, checking the current "Live Adjust Z" is quite easy.
Please follow the path '_calibration > Live Adjust Z_' on the MINI's LCD menu.

For reference, here you can find the structure of the menu.
https://help.prusa3d.com/article/lcd-menu-mini-mini_207531

Please close this issue if you are satisfied with the answer.

Michele Moramarco
Prusa Research

RAMilewski, 2022-06-07:
Thanks Michele!
I missed that!  While it's possible to see the current setting that way, actually getting all the configured Live Z values requires going back and forth between two top level menus eight times.  That's a bit much.  Showing the value to the right of the selected profile on the Settings menu and/or adding a Live Z page to info menu still seems like a good idea.    ...probably not the highest priority for the moment though.  <sigh>

closed 2022-06-07 by RAMilewski