prusa3d/Prusa-Firmware-Buddy #2190: [FEATURE REQUEST] Some ideas to make Prusa Mini even better

HelloItsMeAdm, 2022-06-19:
I have been using **Prusa Mini+** for some time and i had some ideas to make it better.

**1) Disable automatic filament merge when M600**
If you are printing and M600 is called it automatically pulls whole filament from printer. This creates mess on spool or it can create some bigger problems. I know that filament can't be long enought in hotend, but I think that it can pull eg. 10mm and wait for user interaction, he will then make sure to merge back to spool correctly.

**2) Sheet profile when changing filament**
It would be great to bind filaments to their sheets. For example if you are changing from PETG to PLA it would ask user to change his sheet and printer will automatically switch to different sheet profile. This way you will never forget to change it. It would be great to be able to bind it by yourself.

**3) Power panic**
I really don't know why this feature isn't still on Mini, but I really think it would be a nice feature. Also it would be great to make some compability between OctoPrint and Power Loss, either save progress on Raspberry Pi SD card or on Printer's Flash Drive.

**4) OctoPrint controls**
If you have connected Mini to OctoPrint you'll notice that you can't really control printer from LCD display. You can't for example change filament propertly from OctoPrint because if you send M600 to Mini it just parks itselfs and then nothing. Filament change is grayed out in menu, so you have to manually disconnect from OctoPrint, change filament and then connect back to it.
Also you can't eg. change sheet profile do more stuff while you are connected to OctoPrint. I also tried pressing Pause button or even sending G-Code commands to manually make the status to "Paused" but nothing worked.

**5) OctoPrint status**
It would be also great to display current status on Mini when you are printing via OctoPrint. It could be some automatic or with M117 and M73 for Message on display and Print progress.

So these are 5 of my ideas to add/change to Prusa Mini. I hope you like them!

davejenej, 2022-06-19:
They can also finish Prusa Connect (for remote control) and release Wi-Fi adapter for Prusa Mini

Prusa-Support, 2022-06-22:
Hello! Thank you for the ideas and for the detailed post. Some of the features are already in the works, while some others are being evaluated from other requests, all noted from this post and it will be brought to our devs attention! 

Alessandro Pantaleo
Prusa Research

foorschtbar, 2022-10-12:
**6) Sound**
Please add support for the M300 (playing a sound). It would be great if the printer could play a sound when it is done. My printer is in another room and the mini is so quiet that I am not sure if it is still printing or has already finished.

Eldenroot, 2022-10-26:
> **6) Sound** Please add support for the M300 (playing a sound). It would be great if the printer could play a sound when it is done. My printer is in another room and the mini is so quiet that I am not sure if it is still printing or has already finished.

It should play some sound on ERROR or when printing is finished... on maximum volume.

Prusa-Support, 2022-11-24:
Hello, thank you for the requests, reported to our devs! In case of more info from our side, we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

fawkeswei, 2023-08-21:
Sheet profile is a superb idea, please consider adding it 🙏

closed 2024-06-17 by github-actions[bot]